pbatard/rufus #2326: Rufus bricked my whole computer

DiamondBroPlayz, 2023-09-18:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below. (I literally couldn't.)
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it. I COULDN'T

Issue description
-----------------
I got a USB drive and I used rufus to get Windows To Go (Windows 21H2 ISO) running. I chose the "Block Windows To Go from accessing internal drives" option. This was a **HUGE** mistake. After it starting, I got it up and ran it. It wouldn't load. I thought that my USB was too slow and just held down the power button on my desktop. Then, it said that it couldn't access my drive. I tried again to no avail. I then tried to boot into the recovery screen but it was so bricked that not even THAT could run. WHAT DO I DO I CANT WORK ON ANYTHING RELATED TO CODING NOW, AND THAT HAS ALL MY PROJECTS ON IT! PLEASE PLEASE HELP PLEASE I CAN"T FIND ANYTHING ON GOOGLE IM CRYING PLEASE HELP

pbatard, 2023-09-18:
>  I chose the "Block Windows To Go from accessing internal drives" option.

I know this is not what you want to hear but whereas you seem to be under the impression that this setting is something that can be set globally for a computer and that is persisted, __all__ this actually does is carry out step 5 from https://learn.microsoft.com/en-us/previous-versions/windows/it-pro/windows-8.1-and-8/jj721578(v=ws.11) to set the `SanPolicy` environmental variable of the Windows that is being run (so that would be the Windows To Go environment from the USB, __and ONLY the Windows To Go environment from the USB__) to `4` so that it does not attempt to list or access any disk that isn't the current disk that is running Windows.

In other words, the minute you remove the USB drive, the `SanPolicy` settings ceases to apply altogether. It simply cannot be persisted as it is something that Windows reads from the current partition it is trying to boot, and I can assure you that, no, `SanPolicy` does not magically move from one Windows partition or disk to another partition or disk.

If it did, then you can bet that this very public issue tracker (where the project maintainer cannot simply "disappear" issues without a trace, even if they wanted to) would be full of similar reports to yours, where people having created Windows To Go drive with the "Block Windows To Go from accessing internal drives" option would suddenly find that there existing Windows installation can no longer boot once they remove the Windows To Go drive (and just for the record, Rufus is downloaded about 3 million times each month with Windows To Go having been a staple feature of it for years, so, no, it's not a matter of too few people having used the feature to have noticed the issue).

Therefore, we have to conclude that what you experienced was not in fact due to having created a Windows To drive with the "Block Windows To Go from accessing internal drives" option but something else.

Now, with regards to trying to help you, because I am positive that this had nothing to do with Rufus, I can only venture _guesses_, such as the possibility that, because you did attempt to boot a Windows To Go drive, your system may have modified the default boot order and still try to boot the now non existent USB. Therefore, you probably want enter your BIOS/UEFI settings and try to see if you can edit the boot settings to point to the native Windows partition you have on the internal disk or the bootloader that resides there (unfortunately, because these settings are different from computer to computer, nobody can provide exact guidance with regards to which precise settings you need to look for and alter). Or, it is possible that a forced shutdown did have some unexpected effect on your internal disk, in which case your best course of action is to go to a data recovery specialist, so as not to damage your data any further and try to recover what can be recovered.

I am afraid however that, since your issue was not effectively caused by Rufus, and that placing disk offline with Windows To Go is not actually a global computer setting that gets persisted, there is no magical "undo" advice that we can provide to you to restore your existing Windows installation.

DiamondBroPlayz, 2023-09-19:
Alright. Thanks for the help and I’ll try your advice!

closed 2023-09-19 by DiamondBroPlayz