pbatard/rufus #2328: Problème RUFUS

Inoo31, 2023-09-26:
Hello, 

I have tried several times to create a bootable USB key to install Windows on my new PC. Unfortunately, the rufus software always gives me the same error. 

"Failed to assign a volume letter".

Do you have any solutions for me? 

Sincerely
[image]

pbatard, 2023-09-26:
I'm afraid we can't help you unless you post your log, as was requested when you opened the issue.

Inoo31, 2023-09-26:
How can I do that?  Sorry, it's my first time in GitHub :s

pbatard, 2023-09-26:
This is described in the issue creation checklist below, which you __actively removed__ when you opened your issue:

> Checklist
> ---------
> (...)
> - [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
> - [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Oh, and adding an exclamation mark to your issue title will **NOT** help your case.

Inoo31, 2023-09-26:
Sorry, I was just testing the functionality

Inoo31, 2023-09-26:
Rufus x64 v4.2.2074
Windows version: Windows 10 Home x64 (Build 19042.2965)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x040C (fr-FR)
Will use default UI locale 0x040C
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'fr-FR'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 3.0 device 'Verbatim STORE N GO USB Device' (18A5:0243)
1 device found
Disk type: Removable, Disk size: 64 Go, Sector size: 512 bytes
Cylinders: 7685, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0846B21B
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:Checking for Rufus updates...
Checking release channel...

Inoo31, 2023-09-26:
Rufus x64 v4.2.2074
Windows version: Windows 10 Home x64 (Build 19042.2965)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x040C (fr-FR)
Will use default UI locale 0x040C
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'fr-FR'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 3.0 device 'Verbatim STORE N GO USB Device' (18A5:0243)
1 device found
Disk type: Removable, Disk size: 64 Go, Sector size: 512 bytes
Cylinders: 7685, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0846B21B
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:Checking for Rufus updates...
Checking release channel...
Unable to send request: The request has timed out.
Updates: Unable to access version data
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_FR-FR_DV9'
  Detected: Windows 10 ISO (Build 19041.2965)
  Size: 5.7 Go (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_22H2_French_x64v1.iso (5.7 Go)
Selected Windows User Experience options:
• Disable data collection
• Use 'maxen' for local account name

Format operation started
Requesting disk access...
ABORTED: Cannot use an image that is located on the target drive!
Unable to find a drive letter to use

Found USB 3.0 device 'Verbatim STORE N GO USB Device' (18A5:0243)
1 device found
Disk type: Removable, Disk size: 64 Go, Sector size: 512 bytes
Cylinders: 7685, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0846B21B
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 58.9 GB (63208226816 bytes)
  Start Sector: 8064, Boot: No

pbatard, 2023-09-26:
> ``` 
> ABORTED: Cannot use an image that is located on the target drive!
> ```

That's your problem. You cannot have the ISO on the drive you are trying to create because the first thing Rufus needs to do is __completely erase everything on that drive__.

Please move that ISO to your C: drive or somewhere else that is not the USB, and you should find it works better...

Inoo31, 2023-09-26:
I'll try, thanks!

Inoo31, 2023-09-26:
copy ISO files in progress

pbatard, 2023-09-28:
Since you didn't provide an update, I'm going to assume this issue has been resolved.

closed 2023-09-28 by pbatard