pbatard/rufus #2329: Rufus v4.2.2074 x64 crash/hang on USB with Win Xp iso

CDRXavier, 2023-09-27:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [X] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

Rufus frequently crash/hang on the USB device I am using.
The flash drive used is the DT 100 G3, 8GB version.
https://www.kingston.com/en/support/technical/products/dt100g3

I purchased two 8GB version and two 16GB version a while ago.

The ISO is from here:

https://archive.org/details/Windows-XP-Professional-ISO-Version-2002-SP2-OEM-Disc-Image

"""Not official""", but pretty convincing -- I tried burning it to a DVD, and it worked. On like an actual computer (Thinkpad X201 Tablet).

[image]

I tried to just mount the iso and copy the files over to the USB stick, however it didn't work.

Log
---
```
Rufus x64 v4.2.2074 (Portable)
Windows version: Windows 10 Pro x64 (Build 19045.3448)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'AI Mass Storage USB Device' (23A9:EF18)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 981, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x06826BC2
Drive has a Rufus Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 7.5 GB (8068841472 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Checking txtsetup.sif:
  OsLoadOptions = /fastdetect /noguiboot /nodebug
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'KC6BH_K2V4K_9G49Q_TVP2R_4G36W'
  Size: 588.7 MB (Projected)
  Note: File on disk is larger than reported ISO size by 300 KB...
  Uses: WinPE
Using image: VX2POEM_EN_F.ISO (586.9 MB)

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 7.5 GB)
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Cancelling
Format error: Cancelled by user.
\\?\GLOBALROOT\Device\HarddiskVolume26\ was successfully mounted as G:
Re-mounted volume as G: after error

Found USB 2.0 device 'AI Mass Storage USB Device' (23A9:EF18)
Device eliminated because it appears to contain no media
0 devices found
```

CDRXavier, 2023-09-27:
This process also "bricked" my USB stick (by creating a "RAW" partition the size of the entire drive), I have to unplug it, plug it into another computer, and delete the partition.
[image]
(proof that my USB drive works)
I had been using this drive for all sorts of things -- stl files, I plugged it into oscilloscopes to transfer data, etc. *It works*.

pbatard, 2023-09-27:
1. I can't replicate your issue:  
2. I have stopped supporting anything that has to do with Windows XP or Windows 7 and yes that applies even if you run Rufus on a supported platform: https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_XP_Windows_Vista_Windows_7_NO_LONGER_SUPPORTED
3. Somehow, despite the box you checked, I am doubtful that you actually tried on a different computer. If you really did that, youy're going to have to provide more details since, again, I am unable to replicate this issue (and this issue tracker is devoid of other people reporting similar problems).
4. At any rate, this looks like a purely environmental problem, and since this pertains to Windows 7/Windows XP ISOs and I cannot replicate it, I will just close this issue.

closed 2023-09-27 by pbatard

CDRXavier, 2023-09-28:
> 2\. I have stopped supporting anything that has to do with Windows XP or Windows 7

Oh. Well. Maybe adding that to the FAQ would be very helpful.

It might be the environmental variable. I should also try something other than the DataTraveller 100s I have. They don't have the best compatibility (no USB 1.1 support, for instance)

pbatard, 2023-09-28:
Custom MBR ID is 100% focused on older BIOS-oriented images. As a matter of fact, this feature is almost exclusively there for Windows XP installation and nothing else (so much so that I will probably drop it altogether in a future version). There's also the very explicit "Add fixes for old BIOSes" (which I am not planning to remove anytime soon), so stating that _"Rufus isn't focused on older BIOS-oriented images"_ is disingenuous at best.

And the FAQ is pretty explicit about not supporting anything that has to do with Windows 7 and Windows XP, which is very different from saying _"we no longer care about supporting old stuff"_.

Therefore, I would appreciate if you stopped putting words in my mouth that can be demonstrated as 100% not true.