pbatard/rufus #2330: Failing to download OS images through Rufus

HunterAP23, 2023-10-01:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When setting the mode to "Download" and then clicking the button, the loading bar at the bottom immediately completes with the message `Downloading Fido.ps1.lzma` before going back to saying `READY` shortly after. No pop-up window comes up for selecting what OS image to download, and clicking the button again does nothing.

Looking at the log at the bottom, something seems to be happening where the file is successfully downloaded, but the call to PowerShell to run the command fails because it's looking for `{438A6B26-F718-49D0-99E0-E98554CD1E56}.ps1` which obviously doesn't exist. I'm not sure why the file is renamed liked this.

I can confirm that I'm running Windows 11 with PowerShell 7.3.7
Neither of my two USB sticks are appearing as defective after doing a 2-pass scan on each of them.

This issue only happens on the regular version of Rufus 4.2, and works fine on the portable version.

Log
---
```
Rufus x64 v4.2.2074
Windows version: Windows 11 Pro x64 (Build 22621.2361)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'SanDisk Cruzer Glide USB Device' (0781:5575)
Could not get device number for device \\.\R: [0x00000001] Incorrect function.
Could not get device number for device \\.\R: [0x00000001] Incorrect function.
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3740, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0A43A7FE
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 28.7 GB (30764171264 bytes)
  Start Sector: 2048, Boot: Yes
Downloading https://github.com/pbatard/Fido/releases/download/v1.51/Fido.ps1.lzma
File length: 17.4 KB
Successfully downloaded 'Fido.ps1.lzma'
Download signature is valid ✓
Script signature is valid ✓
The argument '.\{438A6B26-F718-49D0-99E0-E98554CD1E56}.ps1' to the -File parameter does not exist. Provide the path to an existing '.ps1' file as an argument to the -File parameter.
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Install the latest PowerShell for new features and improvements! https://aka.ms/PSWindows
Exited download script with code: -196608
```

pbatard, 2023-10-01:
This is most likely due to your system using an unconventional temporary directory (e.g. junction, ramdisk, something else).

For the record, Rufus does download the `.ps1` Fido script under `{<SOME GUID>}.ps1`, and it **attempts** to do so in a temporary directory.

However, if Rufus wasn't able to properly resolve your temporary directory at startup, it falls back to using `.\` (as you can see in the path) instead of a fully qualified path (such as `C:\Users\username\AppData\Local\Temp\`) and this can create issues, since the current directory (`.`) may change during the course of the execution of the application (and yes, that can happen even between the downloading of a script and its execution).

I'm pretty sure that, if you install and run [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview) to capture Rufus' extended log, which contains additional data about the initialisation of the various system directories needed by Rufus, you will see that it produces an early error message about not being able to resolve the temp directory from your system.

And I'm going to be honest here, whereas you may ask why the fall back for the scenario where temp cannot resolve wasn't tested to show that this may cause an issue with some features of the application, the answer to that is that I just don't have the scope, because, a fall back is just that: an __untested__ last ditch attempt at making the application work after detecting unexpected conditions, to try to provide a better alternative than just error out and leave the user stranded.

There are just too many ways in which people can configure a Windows system in a manner that is going to hinder Rufus' ability to perform its operations, and I simply don't have the time to chase after them all. And that is the reason why, if you use nonstandard temp paths, you will run into issues like #2084 and #2296 (which is most likely closely related to your problem).

Now, that is not to say that I'm not trying to address those issues after they are reported (as a matter of fact #2296 is an issue that stems from a fix I applied to try to address the problem reported in #2084), but again, I have to stress out that if your Windows configuration deviates from vanilla in significant ways, you're most likely going to run into problems like this, which application developers are not going to try to pre-empt.

Now, my guess is that the patch I applied in [d5bf53054b58a0f3b6ca5bed64ef9e60da5fff14] to try to address #2296 **may** help your scenario. But to find out if that's likely to be the case, I would need the early output from Rufus that you get in DebugView...

HunterAP23, 2023-10-07:
Thank you for the reply! I wanted to at least confirm that you're spot on with your statement - I am indeed using a non-standard location for the temporary directory. In my case it is a ramdisk that I changed the environment path to point to.

For the time being, using the portable version of Rufus isn't a problem for me, I just figured that I could report the error (and it seems you are already well aware of it).

pbatard, 2023-10-13:
@HunterAP23, can I ask you to please try the BETA for Rufus 4.23, that you can download [here](https://rufus.ie/downloads/) and that includes the changes that I'm hoping should help with your issue?

If not, then I would still like to see your DebugView output to see what it reports.

Thanks.

HunterAP23, 2023-10-16:
I can happily report that the issue is fixed when using the `rufus-4.3_BETA.exe` version of the app. The download script works as intended.

Thank you for the quick fix!

pbatard, 2023-10-16:
Great. Thanks for reporting back. I will close this issue then.

closed 2023-10-16 by pbatard