pbatard/rufus #2331: Feature Request: Make GUI DPI Aware and Resizeable

click-click, 2023-10-01:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
The GUI for Rufus ends up partially popping under the taskbar on my systems. It doesn't seem to be DPI aware and you can't resize the GUI.. It would be a nice enhancement if these items could be implemented,

pbatard, 2023-10-01:
Rufus is DPI aware. But if you are using a low resolution system (or if, say, you're using a 400% factor on a high resolution monitor), the main dialog will never fit.

What is your resolution? And what is your scaling factor?

For what is worth, because it came before, I have no plan to try to accommodate screens where the real estate is too limited. It's just not worth the effort.

click-click, 2023-10-02:
The problem is that the GUI always fills the screen from top to bottom and pops under the taskbar when the drive and format options are expanded. Making the GUI resizable would probably fix those kind of problems.
[image]
[image]

click-click, 2023-10-04:
Not sure if I closed this by mistake or done by the developer. If the latter then I'll notice when it gets closed again and will have to live with it.  Thanks anyway for making Rufus available.

pbatard, 2023-10-04:
You were the one to close it. And I'm afraid that, now, I will close it again, as again, there will always be minimum requirements for applications that are simply not cost effective for developers to try to lower. As far as Rufus is concerned, having a large enough resolution is one of these minimum requirements.

closed 2023-10-04 by pbatard