pbatard/rufus #2332: Unable to patch/boot files for boot

iutarian, 2023-10-02:
Hello there. I'm trying to extract Ghost Spectre Win11 to USB drive by Rufus. I searched for this kind of issues on the web, unfortunately solutions couldn't help me. I tried multiple times, but always the same error: "Unable to patch/boot files for boot". Any help is much appreciated.

iutarian, 2023-10-02:
Rufus x64 v4.2.2074 (AppStore version)
Windows version: Windows 11 Pro x64 (Build 22621.2361)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, _`[`2.06`](url)`_
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found UAS (USB 3.0) device 'APACER Portable HDD UAS Device' (1005:C004)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'SMI USB DISK USB Device' (090C:1000)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 974, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {DF876933-6A30-47B7-A494-D0A36750B54E}
Max parts: 128, Start Offset: 17408, Usable = 8019474944 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {A9E30EFA-5E58-4C20-9992-FE61872EF185}
  Size: 7.5 GB (8018440192 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'WIN11.PRO.22H2.U11.X64'
  Detected: Windows 11 ISO (Build 22621.1702)
  Size: 3.7 GB (Projected)
  Note: File on disk is larger than reported ISO size by 574 KB...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.14.0+)
Using image: WIN11.PRO.22H2.SUPERLITE+SE+COMPACT.U11.(WPE+).ISO (3.7 GB)
Selected Windows User Experience options:
• Bypass SB/TPM/RAM
• Bypass online account requirement

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 7.5 GB)
Waiting for logical drive to reappear...
Formatting to FAT32 (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive2 for exclusive write access
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{02f545e7-6ad4-4fed-ba61-70c9568c2668}\
Successfully remounted \\?\Volume{02f545e7-6ad4-4fed-ba61-70c9568c2668}\ as E:
Extracting files...
Image is a UDF image
[... 1085 lines cut ...]
WARNING: This image appears to be an UNOFFICIAL Windows ISO!
Rufus recommends that you only use OFFICIAL retail Microsoft Windows images, such as
the ones that can be downloaded through the download facility of this application.
Could not mount 'E:\sources\boot.wim [1]' on 'C:\Users\ADMINI~1\AppData\Local\Temp\RufBD55.tmp': [0x00000002] The system cannot find the file specified.
Finalizing, please wait...
E:autorun.inf already exists - keeping it

Found UAS (USB 3.0) device 'APACER Portable HDD UAS Device' (1005:C004)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'SMI USB DISK USB Device' (090C:1000)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 974, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {83D1D3F1-8051-4789-B4C8-28E7E81B099E}
Max parts: 128, Start Offset: 17408, Usable = 8019474944 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {160D2D48-3482-4C03-9981-98301BE7AF4F}
  Size: 7.5 GB (8018440192 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

pbatard, 2023-10-02:
This is an environmental issue with your system.

Judging from your log, the fact that I am not seeing your issue on Windows 11 Pro with the same image as yours, and the fact that you ignored the check list that asked you to test on a different system as well as perform a bad blocks check, I'm going to have to dismiss your issue as purely environmental.

closed 2023-10-02 by pbatard