pbatard/rufus #2335: Attempting to create Windows To Go USB results in bricked drive

jdarn, 2023-10-07:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [X] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Attempting to create a Windows To Go USB has resulted in 2 flash drives being completely bricked. I cannot perform bad block checks, I cannot perform filesystem checks, they are not recognized in the Windows parition manager, they are not recognized as being usable flash media in Linux using GParted, fdisk, or any other such tools. You can see in the logs where the originally dead drive is failing to be read, where I switch to the new flash drive, and where it too fails and from that point forward fails to be recognized as the same device. For reference the Windows 11 install is fresh, it was installed just two days ago. The media being used was just downloaded yesterday from Microsoft and tested to create a normal unmodified Windows installer USB without issue. The flash drives also are new, purchased within the past month, and have already been used to successfully do normal Windows installs, Linux live environments, and Linux installers. It seems like this issue is specific to Windows To Go.

Log
---
Full log was too long for comment, I've attached the file. I'll include what I believe to be the relevant sections in the code blocks below.
```
Rufus x64 v4.2.2074 (Portable)
Windows version: Windows 11 Pro x64 (Build 22621.2361)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found UAS (USB 3.1) device 'WD Game Drive UAS Device' (1058:2653)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 3.0 device 'FC 3281B USB Device' (346D:3281)
Device eliminated because it appears to contain no media
0 devices found
Found UAS (USB 3.1) device 'WD Game Drive UAS Device' (1058:2653)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found UAS (USB 3.1) device 'WD Game Drive UAS Device' (1058:2653)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 3.0 device 'FC 3281B USB Device' (346D:3281)
Device eliminated because it appears to contain no media
0 devices found
Found UAS (USB 3.1) device 'WD Game Drive UAS Device' (1058:2653)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found UAS (USB 3.1) device 'WD Game Drive UAS Device' (1058:2653)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 3.0 device 'FC 3281B USB Device' (346D:3281)
Device eliminated because it appears to contain no media
0 devices found
Found UAS (USB 3.1) device 'WD Game Drive UAS Device' (1058:2653)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 3.0 device 'FC 3281B USB Device' (346D:3281)
[... 383 lines cut ...]
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 3.0 device 'FC 3281B USB Device' (346D:3281)
Device eliminated because it appears to contain no media
0 devices found
Found UAS (USB 3.1) device 'WD Game Drive UAS Device' (1058:2653)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found UAS (USB 3.1) device 'WD Game Drive UAS Device' (1058:2653)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 3.0 device 'FC 3281B USB Device' (346D:3281)
Device eliminated because it appears to contain no media
0 devices found
Found UAS (USB 3.1) device 'WD Game Drive UAS Device' (1058:2653)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Saved C:\Users\jackd\rufus.log
```
[rufus.log](https://github.com/pbatard/rufus/files/12838187/rufus.log)

jdarn, 2023-10-07:
Verified SHA256 hash of the iso manually just now in Windows Terminal: 
`8059A99B8902906A90AFE068AC00465C52588C2BD54F5D9D96C1297F88EF1076`

This matches the hash on the table provided by Microsoft on their downloads page.

pbatard, 2023-10-07:
1. If you check the issue tracker or other places, you will see that you are the **only** person to report this kind of issue.
2. Rufus is downloaded about 3 million times each month, and a not so small fraction of these downloads are being used to create Windows To Go drives
3. A lot of those will have been created using the latest English version of the Windows 11 22H2 ISO that has been released since 2023.05.

__Therefore__, we can safely conclude that the issue is not with the application, because if it was there would be tons of report of "bricked" drives from the thousands of people who have been using Rufus to create a Windows To Go drive with the latest Windows 11 22H2 English ISO release.

Which means that this is an issue with your environment or your hardware.

The fact that it your drives appeared to work okay in some specific scenarios and not with Windows To Go could be due to a number of factors, such as poorly designed drives (since the Windows To Go image application process does a lot of rapid random I/O which may stress a drive a lot more than the usual sequential I/O), the use of "fake" drives (such as an 8 GB masquerading as a 64 GB drive), flaky USB connectivity and so on...

As a matter of fact, since all Rufus really does for the creation of Windows To Go drive, is pretty much follow [these official steps exactly](https://docs.microsoft.com/en-us/previous-versions/windows/it-pro/windows-8.1-and-8/jj721578(v=ws.11)), I am confident that your drives would have failed regardless of whether you used Rufus or not because, once again, the issue is coming from your environment.

Besides, there is absolutely nothing we have control on during the image application phase, where you experience the failure, as Rufus is simply telling the Windows internal API to apply the `sources\install.wim` image to the target drive, and this whole process, __which is controlled exclusively by Microsoft__ is a black box (i.e. Rufus is not copying the files here, it's merely asking Windows to do so and report on the progress).

Finally, you probably want to read [this](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_Rufus_damaged_my_flash_drive).

Now, I appreciate that this is most likely not the answer that you want to read, and that you will want to consider that two drives failing during the same process must be indicative of a problem with the application that initiated that process, but at the very least, the lack of user reports similar to yours, of which there should be plenty if there __really__ was an issue with Rufus, should give you some pause as to whether Rufus is actually to blame...

jdarn, 2023-10-07:
Don't worry about it, I'm not too bothered about the whole thing given they were pretty cheap disposable flash drives. I just wanted to make sure that it wasn't a mistake on my part. Given the type of failure observed here, I'm willing to accept that whatever blackbox process MS uses for Windows To Go was just too much for either the controllers or the NAND flash on these flash drives, since the controllers have effectively gone into dumb mode. 

Thanks for taking the time to reply on the weekend.

closed 2023-10-07 by jdarn