pbatard/rufus #2340: im doing a bootable usb drive for windows 11 im on 97.6 and its been over 5 hours what is this due too

jakcprjf, 2023-10-16:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I have been waiting over 5 hours and its still not finished what could be the reason because people said max it should take 2 hours 

Log
---
```
<FULL LOG>
Rufus x64 v4.2.2074
Windows version: Windows 10 Home x64 (Build 19045.3570)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found UAS (USB 3.0) device 'Seagate Portable UAS Device' (0BC2:2343)
Device eliminated because it was detected as a Hard Drive (score 38 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'Verbatim STORE N GO USB Device' (18A5:0302)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1886, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-GB_DV9'
  Detected: Windows 11 ISO (Build 22621.1702)
  Size: 5.4 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win11_22H2_EnglishInternational_x64v2.iso (5.4 GB)
Selected Windows User Experience options:
• Bypass SB/TPM/RAM
• Bypass online account requirement

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Warning: Could not delete partition(s): [0x80070032] The request is not supported.
Opened \\.\PhysicalDrive2 for shared write access
Requesting logical volume handle...
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
[... 429 lines cut ...]
Extracting: F:\sources\nxquery.sys (53.5 KB)
Extracting: F:\sources\offline.xml (40.9 KB)
Extracting: F:\sources\offlineprofileutils.dll (237.3 KB)
Extracting: F:\sources\oscomps.woa.xml (243.3 KB)
Extracting: F:\sources\oscomps.xml (441.3 KB)
Extracting: F:\sources\osfilter.inf (20.8 KB)
Extracting: F:\sources\outofbox_windows_db.bin (4.7 KB)
Extracting: F:\sources\pnpibs.dll (93.3 KB)
Extracting: F:\sources\pnppropmig.dll (101.4 KB)
Extracting: F:\sources\rdsupgcheck.dll (45.3 KB)
Extracting: F:\sources\reagent.admx (1.2 KB)
Extracting: F:\sources\reagent.dll (581.3 KB)
Extracting: F:\sources\reagent.xml (837 bytes)
Extracting: F:\sources\replacementmanifests\activedirectory-webservices-replacement.man (1.7 KB)
Extracting: F:\sources\replacementmanifests\adsi-router-replacement.man (1.9 KB)
Extracting: F:\sources\replacementmanifests\application-experience-program-compatibility-assistant-replacement.man (11.7 KB)
Extracting: F:\sources\replacementmanifests\appmanmigration-replacement.man (2.1 KB)
Extracting: F:\sources\replacementmanifests\appxalluserstore-replacement.man (2.0 KB)
Extracting: F:\sources\replacementmanifests\appxdeploymentserver-replacement.man (9.8 KB)
Extracting: F:\sources\replacementmanifests\audiommecore-mm-other-migration-replacement.man (1.3 KB)
Extracting: F:\sources\replacementmanifests\authui-migration-replacement.man (8 KB)
Extracting: F:\sources\replacementmanifests\authui-migration-win8-replacement.man (11.3 KB)
Extracting: F:\sources\replacementmanifests\azshci-service-migration-replacement.man (862 bytes)
Extracting: F:\sources\replacementmanifests\bisrv-replacement.man (1.5 KB)
ExExtracting: F:\sources\replacementmanifests\web-services-for-management-core-replacement.man (13.6 KB)

pbatard, 2023-10-16:
Your last line from the log is mangled. Please make sure you don't alter the log while copy/pasting as I am not sure what your last line is and I can't help you unless I can be certain that your log is accurate.

pbatard, 2023-10-19:
Without the clarification I requested, I'm going to have to close this issue.

closed 2023-10-19 by pbatard