pbatard/rufus #2342: Windows 11 features menu not displayed

raknam, 2023-10-17:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I running Windows 11 ARM in Parallel desktop on macOS to execute Rufus and create a Windows 11 (x64) USB Setup Key. ISO in mounted via network because ISO is on my macOS Storage, not the virtual Windows Disk

The problem probably comes from there as the log presents an error
_Could not open image '\\Mac\Downloads\Win11_22H2_EnglishInternational_x64v2.iso': [0x00000057] The parameter is incorrect.
Could not mount Windows ISO for build number detection_

Removed extract part from the log to be allowed to post (65536 char limit). The key is well prepared even I was not able to choose options/features :)

Log
---
```
Rufus x64 v4.2.2074 (Portable)
Notice: Running emulated on arm64 platform
Windows version: Windows 11 Pro arm64 (Build 22000.2295)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3769, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0035B227
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Detected File System: exFAT
  Size: 28.9 GB (31003246592 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3769, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0035B227
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Detected File System: exFAT
  Size: 28.9 GB (31003246592 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
**Could not open image '\\Mac\Downloads\Win11_22H2_EnglishInternational_x64v2.iso': [0x00000057] The parameter is incorrect.
Could not mount Windows ISO for build number detection**
[... 155 lines cut ...]
Total duration: 121.89 milliseconds (121 ms).
NTFS Fixup completed.

Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3769, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {26321263-2E3E-4134-9BC8-358E8226B5AA}
Max parts: 128, Start Offset: 17408, Usable = 31004260864 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {017F018B-070A-4410-B850-9FAC5C61982E}
  Size: 28.9 GB (31002144768 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {8EF52735-C9A1-407B-9EDA-D89D2F391C82}
  Size: 1 MB (1048576 bytes)
  Start Sector: 60553112, Attributes: 0x9000000000000000
```

pineapple63, 2023-10-17:
I think someone else had a similar problem before
It appears Parallels does not properly translate the directory location to the equivalent Windows location, which causes Windows to report the Mac directory location, which then causes the check for the Windows version to fail

pbatard, 2023-10-17:
> I running Windows 11 ARM in Parallel desktop on macOS

There's your issue. And as a result:

> ```
> Could not open image '\\Mac\Downloads\Win11_22H2_EnglishInternational_x64v2.iso': [0x00000057] The parameter is incorrect.
> Could not mount Windows ISO for build number detection
> ```

Being able to obtain the build number is required to display the Windows User Experience dialog.

Rufus requires a real Windows system, not an emulated one. The Mac emulation comes with limitation that prevents the APIs that Rufus relies on from working.

You should therefore report this issue to Apple and let them know that their virtualization __is broken__ because it fails to emulate features that work fine on a real Windows ARM system.

closed 2023-10-17 by pbatard

raknam, 2023-11-02:
Noted :) Thank taking the time to answer