pbatard/rufus #2345: rufus.log in "Rufus" folder

InterClaw, 2023-10-23:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
My report is about the log file itself, rufus.log. It insists on placing itself in a folder called "Rufus". I'm running the portable version. I'm fine with rufus.ini placing itself next to the executable. But why is rufus.log going into a "Rufus" folder?

I thought maybe it behaves this way since it's not running from a folder called "Rufus" in the first place, so I tried that, but then it just created another subfolder also called "Rufus".

This is a bit strange. I propose the log file is not placed in any subfolder, but instead behaves like the .ini file does.

Log
---
```
<FULL LOG>
Rufus x64 v4.2.2074 (Portable)
Windows version: Windows 10 Enterprise x64 (Build 19045.3570)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
0 devices found
```

pbatard, 2023-10-23:
You'll find that, besides its log, Rufus does place a bunch of other files, such as GRUB or Syslinux BIOS bootloaders or DLLs requires for the creation of MS-DOS bootable media into the `Rufus` folder.

So, to answer your question, if we were to place all the files that Rufus may ever produce at the same level as the Rufus executable, when Rufus runs in portable mode, we'd get a bunch of exact opposite reports from yours asking us to please move those files into a subfolder so as not to clog the current folder...

And we just happened to decide that the `rufus.log` was better suited to go into that subfolder. Plus it simplifies the handling between portable and non-portable.

I hope that this answers your question.

closed 2023-10-23 by pbatard

InterClaw, 2023-10-24:
Absolutely! Answered, understood, agreed.