pbatard/rufus #2346: "Remove requirement for an online Microsoft account": will be ignored if "Let's name your device" is skipped

davidhedlund, 2023-10-24:
See https://github.com/ventoy/Ventoy/issues/2630

pbatard, 2023-10-24:
> During the installation, the "Let's name your device" will eventually pop up:

Hmmm, that's not what I have experienced with non Home editions. To your knowledge, is this a Home-specific thing? Have you tested what happens with non-Home versions.

I'll test this, but it may be a while...

> The only workaround for me was to unplug the network cable during the installation. 🫠

That __is__ the expected behaviour for bypassing the requirement for an online account. You are supposed to unplug your network for it to work. If you want the "I don't have internet" button to appear, your computer should obviously not be connected to a network...

So, from where I stand, unless you do see an issue with creating the offline account when your network is unplugged, I don't really see much to do in Rufus. If you want an offline account, you must be offline during setup, as this is how offline account creation has worked for quite some time (and the goal of the  "Remove requirement for an online Microsoft account" option is to make Windows 11 behave like Windows 10 in that respect.

davidhedlund, 2023-10-27:
> is this a Home-specific thing?

Yes.

> That **is** the expected behaviour for bypassing the requirement for an online account.

I updated the top post. Thank you very much for pointing this out!

davidhedlund, 2023-10-27:
>  is this a Home-specific thing?

When I performed the installation without internet connected as you said, the prompt was instead "Who's going to use this device" and it's mandatory to fill it, but you will not be forced to sign in. So everything is in order.

> That **is** the expected behaviour for bypassing the requirement for an online account.

I updated the top post. Thank you very much for pointing this out!

davidhedlund, 2023-10-27:
After watching https://youtu.be/EOUcvgqOV-0?t=284 I'm a bit confused. Is it only required by Rufus that the Ethernet cable must be unplugged? If you install it on a laptop with inbuilt wireless NIC and have a wireless router in the same room that is connected to the internet: I guess Rufus will bypass internet detection in the Windows installer. Or do you have to disable the wireless network in this case, or move the laptop somewhere where wireless network is not available? @pbatard 

If only the Ethernet cable must be unplugged, then I guess we should change the title to: _Extend "Remove requirement for an online Microsoft account" with " (requires that the network cable is unplugged before the computer is started)"_

pbatard, 2023-10-27:
First of all, please do __NOT__ edit your post to remove data that was related to previous answer, as it makes it __INCREDIBLY DIFFICULT FOR OTHER PEOPLE IN THE SAME SITUATION AS YOU WERE TO UNDERSTAND WHAT'S BEEN GOING ON!__

>  Is it only required by Rufus that the Ethernet cable must be unplugged?

No. This was required for Windows 10 and earlier Windows 11 installation, regardless of the utility you use to create the media.
Microsoft changed the behaviour in Windows 11 22H2, and made it so that, even if you didn't have a network connection when being prompted for account creation, you would not get the _"I don't have internet"_ button and therefore could not create an offline account. So the requirement for network to be unavailable to see the the _"I don't have internet"_ button has always been there (which makes sense) and all Rufus does is re-enable that button to be displayed on later versions of Windows 11.

> If you install it on a laptop with inbuilt wireless NIC and have a wireless router in the same room that is connected to the internet

Well, the expectation is that your network __will__ be secured, and therefore require a password to connect and therefore that you will __NOT__ have automatic network connectivity to it. So you should not have to disable wireless network.

Now, if you use a wireless network without any security and let WIndows connect to it automatically, that is your problem, because it would not be a typical situation.

And again, since we are restoring behaviour that is aimed at offline account creation, and that therefore implicitly requires the network not to be connected, I am not planning to over-encumber a straightforward list with the various caveats and whatnot of each option, especially as, if you do want a local account, you can just use the option where Rufus will create it for you (and please also do not attach screenshots from __OLD__ versions of Rufus, that are no longer relevant, as the _"Set a local account using the same name as this user's"_ has long been replaced and, if you want support, you should not be using old versions of Rufus.

davidhedlund, 2023-10-27:
> First of all, please do **NOT** edit your post to remove data that was related to previous answer, as it makes it **INCREDIBLY DIFFICULT FOR OTHER PEOPLE IN THE SAME SITUATION AS YOU TO UNDERSTAND WHAT'S BEEN GOING ON!**
> 
> > Is it only required by Rufus that the Ethernet cable must be unplugged?
> 
> No. This was required for Windows 10 and earlier Windows 11 installation, regardless of the utility you use to create the media. Microsoft changed the behaviour in Windows 11 22H2, and made it so that, even if you didn't have a network connection when being prompted for account creation, you would not get the _"I don't have internet"_ button and therefore could not create an offline account. So the requirement for network to be unavailable to see the the _"I don't have internet"_ button has always been there (which makes sense) and all Rufus does is re-enable that button to be displayed on later versions of Windows 11.
> 
> > If you install it on a laptop with inbuilt wireless NIC and have a wireless router in the same room that is connected to the internet
> 
> Well, the expectation is that your network **will** be secured, and therefore require a password to connect and therefore that you will **NOT** have automatic network connectivity to it. So you should not have to disable wireless network.
> 
> Now, if you use a wireless network without any security and let WIndows connect to it automatically, that is your problem, because it would not be a typical situation.
> 
> And again, since we are restoring behaviour that is aimed at offline account creation, and that therefore implicitly requires the network not to be connected, I am not planning to over-encumber a straightforward list with the various caveats and whatnot of each option, especially as, if you do want a local account, you can just use the option where Rufus will create it for you (and please also do not attach screenshots from **OLD** versions of Rufus, that are no longer relevant, as the _"Set a local account using the same name as this user's"_ has long been replaced and, if you want support, you should not be using old versions of Rufus.

Firstly, thank you very much for the detailed answer. You helped me to better understand how all of this is interconnected.

> I am not planning to over-encumber a straightforward list with the various caveats and whatnot of each option

I changed the title again to reflect your answers. For the sake of the latest Windows version and release, don't you agree that it would be useful to change: "Remove requirement for an online Microsoft account" to "Remove requirement for an online Microsoft account (requires that the network is not publicly available before the computer is started)"?

pbatard, 2023-10-27:
No, I don't agree. I explained why.

closed 2023-10-27 by pbatard

davidhedlund, 2023-10-27:
> No, I don't agree. I explained why.

I respect your decision.