pbatard/rufus #2347: QUIC Question:What do you recommend to use in Linux/Manjaro?

Qodestackr, 2023-10-25:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2023-10-25:
This is not really the place to ask such questions, and you will find that, unless you weigh the pros and cons of each distro, what one person might recommend may be completely ill suited to you. It __really__ depends of your prior knowledge of UNIX/Linux, what you plan to do with Linux and your level of skills.

As such, I will not recommend anything to you.

closed 2023-10-25 by pbatard

fanfan54old, 2023-10-26:
Hi,
Just in case this helps,
Maybe @Qodestackr asked for an alternative to Rufus on Linux...
Maybe you can find alternatives and software compared between each other with a simple Google search, but I just want to say that, the last time I had to flash a bootable ISO (for a live USB I think 🙃) to my USB flash drive **on Linux (Ubuntu)**, I think I used [mkusb](https://help.ubuntu.com/community/mkusb) and it worked flawlessly for me with its simple UI, I think it's made with Zenity... sorry if that's off-topic 🙃
It's been a long time and I don't know if it works OK on other Linux distros but maybe you can try it 🙃

Toni500github, 2023-11-10:
Use `cat`. Works on every Linux distro and it's way faster 

`cat file.iso > /dev/[your USB drive e.g sdb]`

`cp` works too

`cp file.iso /dev/[your USB drive e.g sdb]`

Qodestackr, 2023-11-12:
Thank ya'll I have an Ubuntu now :upside_down_face: