pbatard/rufus #2348: obsd

babut85, 2023-10-26:
is it possible to add obsd support? they changed something and now cd images are not opened via rufus(but are opened by other utilities)

﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2023-10-26:
> is it possible to add obsd support? 

Is it possible for you to provide a log __as explicitly requested__?

babut85, 2023-10-26:
Rufus x64 v4.3.2090
Windows version: Windows 10 Enterprise LTSB x64 (Build 19044.3570)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'EPSON Storage USB Device' (04B8:1121)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'General Flash Disk3.0 USB Device' (346D:5678)
1 device found
No volume information for drive 0x84
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7648, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 0
Disk GUID: {C0CB7B1A-5981-4524-B6E5-1D796BFFD615}
Max parts: 128, Start Offset: 17408, Usable = 62914525696 bytes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'OpenBSD/amd64   7.4 Install CD'
  Size: 603.4 MB (Projected)

pbatard, 2023-10-26:
https://github.com/pbatard/rufus/wiki/FAQ#openbsd

closed 2023-10-26 by pbatard