pbatard/rufus #2350: [feature request]: Windows User Experience: Disable “Let’s Finish Setting up Your Device”

davidhedlund, 2023-10-28:
"After installing the Windows 11/10 update or creating a new user account, you may see the “Let’s finish setting up your device” message. It’s annoying." - https://www.minitool.com/news/disable-lets-finish-setting-up-your-device-on-windows-11-10.html

pbatard, 2023-10-28:
>  It’s annoying

Yes, it's annoying to __you__.

Sadly, I am not designing Rufus to remove inconsequential elements that appear to annoy you.

pbatard, 2023-10-28:
For what is worth, my goal for WUE is to disable the annoyances that prevent streamlining the installation process in order to get to an installed version of Windows you can start to interract with.

What you describe appears to happen after that, so it's out of scope of WUE.

closed 2023-10-28 by pbatard

davidhedlund, 2023-10-28:
> For what is worth, my goal for WUE is to disable the annoyances that prevent streamlining the installation process in order to get to an installed version of Windows you can start to interract with.
> 
> What you describe appears to happen after that, so it's out of scope of WUE.

I understand.

davidhedlund, 2023-11-11:
> For what is worth, my goal for WUE is to disable the annoyances that prevent streamlining the installation process in order to get to an installed version of Windows you can start to interract with.
> 
> What you describe appears to happen after that, so it's out of scope of WUE.

I understand your goal, so are there absolutely no edge cases that are worth consideration even after the installation process? For example:

"As soon as it boots, and without asking any permission, [Windows 11 starts to send data to online servers](https://www.techspot.com/news/97535-windows-11-spyware-machine-out-users-control.html). The user's personal details, location or hardware information are reported to Microsoft and other companies to be used as telemetry data. All of this is done is the background, and users have no easy way to prevent it—unless they switch the computer offline." - https://www.gnu.org/proprietary/proprietary-surveillance.en.html#SpywareInWindows

To disable telemetry, see the "Use Registry Editor" section at https://windowsreport.com/disable-windows-11-telemetry/

pbatard, 2023-11-11:
Well, if you have just installed Windows, there won't be that much history to send, and, as you should know, the way Rufus disables the online account requirement is by using a bypass that __Microsoft themselves added__ and that they could remove if they really wanted to, so the idea that I'd somehow be interested in concealing whether Windows was installed using a local account is a bit risible, because we're not doing anything that Microsoft themselves does not officially provide provision for in the first place. There are a lot of people who seem to be under the impression that somehow Rufus "hacks" into Windows to perform operations that Microsoft never allowed, but the truth of the matter is, we absolutely play on the field and with the rules that Microsoft provides because, as it happens, Microsoft does have a lot of corporate customers that don't want online accounts or other stuff, so they kind of have to leave some ways for these to be disabled... and that is exactly what Rufus uses.

So, once again, we me be absolutely clear: My one and only goal with WUE is to streamline the installation process so that you can launch an install, go for a coffee, and come back to an installed version of Windows, that is fully installed and that you can start to interact with. So, unless the installation process stops to ask if you agree to send telemetry, I'm not planning to do anything about it, especially if, as it happens, the telemetry you describe can be disabled through an __easily__ accessible setting once Windows is installed.

davidhedlund, 2023-11-12:
> So, once again, we me be absolutely clear: My one and only goal with WUE is to streamline the installation process so that you can launch an install, go for a coffee, and come back to an installed [...]

I suggest that you mention this in the https://github.com/pbatard/rufus/wiki/FAQ so you can refer to it.

I requested the Free Software Foundation to add Rufus to https://www.fsf.org/news/lifes-better-together-when-you-avoid-windows-11 :

_If you absolutely must use Windows 11, it seems that the free utility [Rufus](https://en.wikipedia.org/wiki/Rufus_(software)) can bypass this requirement._