pbatard/rufus #2353: Feature request: option to simply skip copying boot.wim and install.wim to the USB drive

RadarNyan, 2023-11-01:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I'm aware that there have been many issues related to this (I'm not going to mention them by number to avoid sending notifications to people who commented on those very old issues), and I am aware of issue no. 322. However, I'd still like to suggest that you add this option to simply skip copying boot.wim and install.wim files, which I believe would be very easy to implement.

With this option, it makes it easier (and faster) for users who want to make customizations to those files, there's no need to wait for Rufus to copy the files, and there's no need to delete them before manually coping the new ones in. If someone insists on using FAT32 for UEFI boot, they can perform the splitting of the install.wim manually and copy them to the drive after Rufus finishes the job.

Log
---
```
(not applicable)
```

pbatard, 2023-11-01:
This is sysadmin/power-user stuff, so see https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_cant_you_just_add_the_feature_I_want.

I just don't see enough people benefiting from this.

&rarr; Rejected.

pbatard, 2023-11-01:
For the record, Rufus already provides an option to create a blank UEFI:NTFS drive. So all you have to do to accomplish what you want (for UEFI system installation) is copy the files you want after create a blank UEFI:NTFS drive with Rufus, and you can easily use Windows ISO mounting facility or 7-zip to copy only the ISO content you want.

closed 2023-11-01 by pbatard