pbatard/rufus #2354: latest win 11 23h2 customization

rajkumarindia, 2023-11-01:
Rufus 4.3 with win 11 23h2 official iso not prompting for customization.

Like bypassing 4gb limit and TPM requirements.

saintyuk, 2023-11-01:
Seems fine for me? Any further details?
[image]

pbatard, 2023-11-01:
@rajkumarindia, please provide your log, or I will close this issue.

Dook1, 2023-11-03:
I have the same issue.
[rufus.log](https://github.com/pbatard/rufus/files/13246067/rufus.log)

pbatard, 2023-11-03:
I'm pretty sure the issue has nothing to do with using 23H2 and will happen with every recent Windows ISO.

@Dook1, your issue is that the __Windows__ ISO mounting facility (because for version detection, Rufus does mount the ISO itself but instead it asks Windows to do it) doesn't seem to be able to mount your `G:\Microsoft Windows 11, version 23H2, build 22631.2428 (updated October 2023) - Оригинальные образы от Microsoft MSDN [Ru]\ru-ru_windows_11_business_editions_version_23h2_x64_dvd_bcc01258.iso` image.

This is a Windows internal issue, not a Rufus issue, and as such, it is yours to fix.

Foe one thing, I'd try moving the ISO to the C: drive and rename it to something shorter like `Windows 11 23H2.iso`.

netcore-jroger, 2023-11-09:
I have same problem.

CPU: i7 6700HQ
OS: Win10 Pro 22H2

pbatard, 2023-11-09:
@netcore-jroger, please provide your log or I will ignore your issue.

netcore-jroger, 2023-11-09:
Sorry, I don't use it often, I forgot. Need to click the start button to get that dialog box to pop up.

rajkumarindia, 2023-11-09:
Im grateful to you for taking up the issue.

Im out of station .

Will  send tomorrow.

Thank you  and wish you all the best

nbartley, 2023-11-13:
The bypass still works, you just have to not enable a network connection during setup. If you do, the bypass fails.

pbatard, 2023-11-13:
No log &rarr; Closed.

closed 2023-11-13 by pbatard