pbatard/rufus #2356: A lil heads up

LoudMouthJester, 2023-11-11:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

The download for Rufus-2.15_arm.exe on rufus.ie/downloads flagged 'This file contains a virus or malware.'
Log
---
[image]

```
<FULL LOG>
```

pbatard, 2023-11-11:
https://github.com/pbatard/rufus/wiki/FAQ#user-content-Antivirus_X_reports_that_Rufus_contains_malware

closed 2023-11-11 by pbatard

LoudMouthJester, 2023-11-22:
No doubt. Had no idea. Thanks for the link. Much appreciated