pbatard/rufus #2360: Rufus 4.3 - Strings untranslated

bovirus, 2023-11-15:
@pbatard 

Issue description
-----------------

Seems that not all strings are are ranslated (also if .po file is 100%) translated.

Example of strings untranslated in official version 4.3.

[image]

Strings

MSG_337 "An additional file ('diskcopy.dll') must be downloaded from Microsoft to install MS-DOS (...)"
MSG_338 "Revoked UEFI bootloader detected"
MSG_339 "Rufus detected that the ISO you have selected contains a UEFI bootloader that has been revoked (...)"
MSG_340 "a \"Security Violation\" screen"
MSG_341 "a Windows Recovery Screen (BSOD) with '%s'"
MSG_342 "Compressed VHDX Image"
MSG_343 "Uncompressed VHD Image"
MSG_344 "Full Flash Update Image"
MSG_345 "Some additional data must be downloaded from Microsoft to use this functionality (...)"

are not included in language file also in official build 4.3.

If I check the screenshot seems that string about bootloade revoked is longer then string that appear in teh screen.

Please check .loc file and .po file.
Please include all strings untranslated in .po/.loc files process.

Thanks.
```

bovirus, 2023-11-15:
@pbatard 

Also translating all new strings remain "Error code:" untranslated

[image]

pbatard, 2023-11-15:
Yes, that is intentional, as I only issue a round of translation about once a year, on account that, as I believe I told you already, it is very demanding to try to follow up with about 35 translators, so I need to be able to allocate enough time to bring all the translation up to date. 

Basically, this requires about 1~2 week full time work for me, knowing that I also need to provide about 1 month for translators to translate the new data. The new messages you mention will be translated when I start the next translation round, but no sooner.

As to `Error code` not being translated, this is because this is a message that will be generated in English by the bootloader, so this is what non English speakers will see when they attempt to boot, and therefore should be left untranslated.

pbatard, 2023-11-15:
See also: https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_dont_you_accept_unsolicited_translation_updates

closed 2023-11-15 by pbatard

bovirus, 2023-11-15:
@pbatard 

Thanks for teh info.

For me to have a program with a mix of strings english and italian doens't make a good feel.
For me you can update english file and pollock tool.
The rest of job are in charge of translators.
Then step by step the translator that will update the language file found their languages in a new build.
For me I temporary fixed langauge issues using and exeternal .loc language file.
I don't like to have English message in a sofwtare with UI in Italian.
Thanks.

pbatard, 2023-11-15:
I explained why what you are requesting is not feasible. In order to provide quality software, I do validate translation updates that are submitted to me, especially if they have to do with new features, and I simply can't do that, piecemeal, without multiplying the amount of time I need to spend dealing with translation updates, which is already more considerable than people imagine, by 3 or 4.

You are under the __false__ impression that all that's needed to make a translation update in Rufus is pick whatever the translator submitted, assume that it's all going to be fine, and shove it there without a second glance. And you are also assuming that translators will provide timely updates on a regular basis, as well as follow the project on their own to check whether new elements require translation (rather than me having to e-mail 35 translators and then chase after the ones who haven't submitted their updates). Finally, you are also assuming that, during the course of introducing a new feature requiring a translation, I am not going to alter it, thereby requiring translators to waste their time translating something one way, and then a different way.

In short, what you are requesting only works in an ideal fictional world, but would be exceedingly time consuming in reality, and would detract to the overall health of the project. As such, it is simply not feasible, and I must admit I am growing tired of having to explain to you, time and time again, why the choices I have to make in the context of Rufus have actually been very deliberately pondered over the course of years and stem directly from both logic and from __actually__ trying to provide the best software possible in the context of resources that are not unlimited and the world we are working in, as well as the people inhabiting in it, not behaving in an ideal manner.