pbatard/rufus #2361: Windows To Go - ISO Image Extraction Failure (The device is used by another process)

flyingtanker, 2023-11-16:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Same as previous [one](https://github.com/pbatard/rufus/issues/1637), ssd with enclosure in my case. Neither latest rufus version or 3.13 or 3.11 worked for me.
Image was downloaded directly from microsoft.
Here will be couple of screenshots
[image]
[image]

Log
---
```
[rufus.log](https://github.com/pbatard/rufus/files/13372116/rufus.log)
Log was too big thus a file
```

pbatard, 2023-11-17:
This doesn't look like a Rufus issue.

The way Rufus creates a Windows To Go drive is that:
1. It asks Windows to mount the ISO as a virtual CDROM drive
2. It asks Windows to apply the `install.wim` image found on the mounted ISO.

None of these operations are under the direct control of Windows, so, if they fail, then it's an issue with Microsoft Windows or the hardware rather than us.

Especially, it looks like you might have an have an issue with a flaky SSD enclosure because, by the looks of it, the device disappears from the system altogether as Windows is still extracting WIM files onto it, hence the error you get.

You will therefore have to figure out the software or hardware causes that can make your device disappear, which, I'm afraid, not much of anybody else will be able to help you with, as it looks like a pure environmental issue.

pbatard, 2023-11-23:
Considering that there has been no follow up, I will close this issue.

closed 2023-11-23 by pbatard