pbatard/rufus #2365: Issue with Rufus Turning USBs to RAW File System

mabdullahh, 2023-11-20:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<

Hey there,

I've run into a bit of a snag with Rufus, and I'm hoping you can help me out. Usually, I've had smooth experiences using Rufus for creating bootable USBs for Windows installations. However, recently, when I tried making a Windows 10 bootable media for a UEFI based computer, things went south.

The USBs that have been my go-to with Rufus are suddenly turning into RAW file systems, and I can't figure out why. I've used Rufus many times before without any problems, so this has caught me off guard. I even tried reformatting the USBs using Rufus again, but it doesn't seem to be working.

I've also given a shot at using regular tools on my computer, along with some third-party partition wizard tools and Diskpart, but none of them are resolving the issue. I attached a file with more details if that helps.
P.S I've used three pen drives to re produce this log file.

Any insights or suggestions you have would be much appreciated. Thanks!
>

Log
---
```
Rufus x64 v4.3.2090
Windows version: Windows 10 Pro x64 (Build 19045.3754)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'Generic USB SD Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
NOTE: This device is a USB 3.0 device operating at lower speed...
Found USB 2.0 device 'Generic USB SM Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB MS Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic Mass Storage USB Device' (1908:9999)
Found USB 2.0 device 'Generic USB CF Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
3 devices found
No volume information for drive 0x88
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1881, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {DC81B970-CCB3-4260-94AF-33500E4A3D30}
Max parts: 128, Start Offset: 17408, Usable = 15472012800 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: (Unrecognized)
[... 184 lines cut ...]
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB SM Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic USB MS Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic Mass Storage USB Device' (1908:9999)
Found USB 2.0 device 'Generic USB CF Reader USB Device' (058F:6362)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
2 devices found
No volume information for drive 0x82
Disk type: Removable, Disk size: 67 GB, Sector size: 512 bytes
Cylinders: 8158, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x002963F2
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 62.5 GB (67107815424 bytes)
  Start Sector: 2048, Boot: Yes
Closed Windows format prompt
```

pbatard, 2023-11-20:
> The USBs that have been my go-to with Rufus are suddenly turning into RAW file systems, and I can't figure out why.

Power issue on your USB bus? Flaky USB connection?

All signs point to a hardware issue on your side, so I will ask you to answer truthfully:

Did you really test on a different computer, as you indicate you did in the check-list?
And did you try to run a bad blocks check, as you indicate you did in the check-list?

These 2 tests, along with trying different USB ports, are really the ones you want to check, so I need to know that you did not just tick the checkboxes because you were annoyed at leaving them blank, and actually performed these tests.

pbatard, 2023-11-28:
No update &rarr; Closed.

closed 2023-11-28 by pbatard