pbatard/rufus #2370: WMI

OvalPiston, 2023-11-28:
Issue description
-----------------
I'm getting a 'WIM processing error:' message int he log. Consequently, the Status after running shows 'Failed'. I've tried two different USB drives in two different computers and I get the same result. 

When I click 'Start', I keep the 'Customize Windows installation' options already checked:
Remove 4GB+ requirement
Remove online account
Set regional options
Disable data collection
Disable BitLocker

Log
---
```
Rufus x64 v4.3.2090
Windows version: Windows 10 Pro for Workstations x64 (Build 19045.3693)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 3.0 device 'SanDisk Extreme Pro USB Device' (0781:5588)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0xC4C04623
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 119.2 GB (128041578496 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI System Partition (0xef)
  Detected File System: FAT12
  Size: 1 MB (1048576 bytes)
  Start Sector: 250083256, Boot: No
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CPBA_X64FRE_EN-US_DV9'
  Detected: Windows 11 ISO (Build 22621.2428)
  Size: 6.1 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: en-us_windows_11_business_editions_version_23h2_x64_dvd_a9092734.iso (6.1 GB)
Found USB 3.0 device 'SanDisk Extreme Pro USB Device' (0781:5588)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0xC4C04623
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 119.2 GB (128041578496 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI System Partition (0xef)
  Detected File System: FAT12
  Size: 1 MB (1048576 bytes)
  Start Sector: 250083256, Boot: No
Found USB 3.0 device 'SanDisk Extreme Pro USB Device' (0781:5588)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0xC4C04623
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 119.2 GB (128041578496 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI System Partition (0xef)
  Detected File System: FAT12
  Size: 1 MB (1048576 bytes)
  Start Sector: 250083256, Boot: No
Checking for Rufus updates...
Checking release channel...
No new release version found.
Found USB 3.0 device 'SanDisk Extreme Pro USB Device' (0781:5588)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0xC4C04623
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 119.2 GB (128041578496 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI System Partition (0xef)
  Detected File System: FAT12
  Size: 1 MB (1048576 bytes)
  Start Sector: 250083256, Boot: No
Selected Windows User Experience options:
• Bypass SB/TPM/RAM
• Bypass online account requirement
• Disable data collection
• Use the same regional options as this user's
• Disable bitlocker

Format operation started
Requesting disk access...
Will use 'I:' as volume mountpoint
Opened \\.\PhysicalDrive5 for exclusive write access
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 119.2 GB)
● Creating UEFI:NTFS Partition (offset: 128042627072, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume \\?\Volume{36cea1da-8dfa-11ee-9feb-14358b029bad}\
Disabling file indexing...
Opened \\?\Volume{36cea1da-8dfa-11ee-9feb-14358b029bad} for exclusive write access
Writing Partition Boot Record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted \\?\Volume{36cea1da-8dfa-11ee-9feb-14358b029bad}\ as I:
Extracting files...
Image is a UDF image
Extracting: I:\autorun.inf (128 bytes)
Extracting: I:\boot\bcd (16 KB)
Extracting: I:\support\logging\winsetupetw.dll (73.5 KB)
Applying Windows customization:
Created 'I:\sources\appraiserres.dll' placeholder
Mounting 'I:\sources\boot.wim[2]'...
WIM processing error: D:\temp\Ruf7B7E.tmp\Windows\WinSxS\amd64_microsoft-windows-registry-editor_31bf3856ad364e35_10.0.22621.1_none_55c48118cf96d6d6\Registry Editor.lnk [[0x00000032] The request is not supported.]
Could not mount 'I:\sources\boot.wim [2]' on 'D:\temp\Ruf7B7E.tmp': [0x00000032] The request is not supported.
Finalizing, please wait...
I:autorun.inf already exists - keeping it
NTFS Fixup (Checkdisk)...

pbatard, 2023-11-28:
Issue is with your environment.

I am seeing no problem doing the same operation as you do on Windows 11 Pro for Workstations x64, using `en-us_windows_11_business_editions_version_23h2_x64_dvd_a9092734.iso`:

<details>
  <summary>Click the arrow to see my log:</summary>

```
[Tue Nov 28 15:22:45 2023]
Rufus x64 v4.3.2090
Windows version: Windows 11 Pro for Workstations x64 (Build 22631.2715)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB device 'SanDisk Extreme USB Device' (0781:5580) [ID]
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 29.2 GB (31405899776 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CPBA_X64FRE_EN-US_DV9'
  Detected: Windows 11 ISO (Build 22621.2428)
  Size: 6.1 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: en-us_windows_11_business_editions_version_23h2_x64_dvd_a9092734.iso (6.1 GB)
Selected Windows User Experience options:
• Bypass SB/TPM/RAM
• Bypass online account requirement
• Disable data collection
• Use the same regional options as this user's
• Disable bitlocker

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive9 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 29.2 GB)
● Creating UEFI:NTFS Partition (offset: 31405842432, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
[... 1013 lines cut ...]
Found USB device 'SanDisk Extreme USB Device' (0781:5580) [ID]
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {512C1E63-F04B-410A-8177-F7261537A0BE}
Max parts: 128, Start Offset: 17408, Usable = 31406914048 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {828EFF0E-321F-46DD-A94F-9AC77AA4C91D}
  Size: 29.2 GB (31404793856 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {EC50E5FE-51A9-4BC9-B0A7-06008280AF9B}
  Size: 1 MB (1048576 bytes)
  Start Sector: 61339536, Attributes: 0x9000000000000000
```
</details>

Rufus does rely on Windows APIs to mount and process the WIM file, so, if that doesn't work and if nobody else is reporting a similar problem, it's a Windows issue with your environment that you will have to troubleshoot on your own.

closed 2023-11-28 by pbatard

OvalPiston, 2023-11-28:
I had to disable all of the helpful features, and then it worked: https://revi.cc/docs/faq/during/rufus/

Remove 4GB+ requirement
Remove online account
Set regional options
Disable data collection
Disable BitLocker

pbatard, 2023-11-28:
Well, yes, if you disable the _Bypass SB/TPM/RAM_ feature, it should work because that is the feature that requires mounting the WIM image to alter the registry.

OvalPiston, 2023-11-28:
I'm not assuming everybody will run into this issue. Thanks for your quick responses.