pbatard/rufus #2374: Rufus not answering when USB plugged

ZaZaMant, 2023-12-02:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x ] I also tried one or more of the following:
  - [ x] Using a different USB drive.
  - [x ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
(Im french, feel free to correct any basic error you notice in my text)

That's literaly every thing in the title, I m trying to put ubuntu as my main OS on my portable computer. I had a bootable usb key before, and when I plugged it and tried to install it. It got stuck somewhere, I didn't understood and decided to format the key and do it all over again (because it was a bootable key made by a friend). So I downloaded ubuntu and rufus, but my rufus literaly wouldn't open. I figured that when I don't have any USB card plugged in rufus opens, and if I plug it when rufus is opened, it doesn't respond and must be closed.
So I m a bit lost, I also never heard about Sophus, I have avira and I deactivated it during some tests (maybe I missed some, I will try again later)

Log
---
```
<FULL LOG>

Rufus x64 v4.3.2090 (Portable)
Windows version: Windows 11 Home x64 (Build 22621.2715)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x040C (fr-FR)
Will use default UI locale 0x040C
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'fr-FR'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
0 devices found

```

pbatard, 2023-12-02:
Looks like your USB key might be dead. The fact that _"It got stuck somewhere"_ when you tried to use it during Ubuntu installation would tend to indicate that it was already failing then...

Is the USB key accessible in Windows at all? Can you format it using Windows explorer?
If not, then it's definitely a hardware issue with your USB key.

Also, you state that you tried to use a different USB drive. Is that true? From your report it appears that you only have an issue with one USB key, not two.

ZaZaMant, 2023-12-02:
Yeah I can access files in the USB key, I formated it in the file manager, and about the second USB key I didn't try to install ubuntu on both of them, I checked if rufus closed if it wasn't the same USB key.

pbatard, 2023-12-02:
Then you are probably using a security solution that is preventing/slowing access to the drive. What happens if you wait for 1 hour? Does Rufus eventually open?

I would also appreciate if you could perform the following:
1. __Without the USB key plugged in__ launch Rufus and press <kbd>Alt</kbd>-<kbd>.</kbd>. You should see a message stating `USB Debug enabled` (or `Debug USB activé` in French) on Rufus status bar at the bottom. Note that if you are using a French keyboard, it is possible that the <kbd>.</kbd> may be mapped somewhere else).
2. Close Rufus and plug your USB key
3. Install and run DebugView from https://learn.microsoft.com/en-us/sysinternals/downloads/debugview. NB: If you're not sure which `.exe` to run, you should try `dbgview64.exe`.
4. With DebugView running and your USB key plugged in, launch Rufus. You should see some message starting with `[####] *** Rufus init ***`
5. Copy the output from DebugView and post it here.

pbatard, 2023-12-02:
Thanks. The last line from the log is from a different application (most likely Steam) and is irrelevant, but by the looks of it, a third party application or security solution is preventing access to the USB drives. This is something we've seen with Sophos products and I'm pretty sure you won't experience that issue on a __vanilla__ installation of Windows (i.e. a Windows system that has been freshly installed, with not additional applications).

Unfortunately, outside of trying to wait to see if whatever application is interfering is simply slowing down access, I don't really have a solution for you, because this is clearly an environmental problem, and the only other similar issue that was ever reported was from people running Sophos Antivirus. Your current system is simply incompatible with Rufus.

pbatard, 2023-12-09:
Since there is nothing else I can add on this issue, which appears to be purely environmental, I will close it.

closed 2023-12-09 by pbatard