pbatard/rufus #2380: Very minor logging issue

sabaasa, 2023-12-08:
﻿

Checklist
---------
- [+] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ + dont see anything in open issues list] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [+] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [+] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus is unable to detect Windows 10 IoT Enterprise LTSC 2021 edition and just says "Unknown Edition 0xBF", should be an easy enough fix.

Log
---
```
Rufus x64 v4.3.2090 (Portable)
Windows version: Windows 10 (Unknown Edition 0xBF) x64 (Build 19044.3693)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'ADATA USB Flash Drive USB Device' (125F:DD24)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3773, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x732B46A5
Drive has an unknown Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Detected File System: exFAT
  Size: 28.9 GB (31003246592 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: EFI System Partition (0xef)
  Detected File System: FAT16
  Size: 32 MB (33554432 bytes)
  Start Sector: 60555264, Boot: No
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCLA_X64FREO_EN-US_DV9'
  Detected: Windows 11 ISO (Build 22621.2428)
  Size: 5.7 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: X23-59465_22631.2428.231001-0608.23H2_NI_RELEASE_SVC_REFRESH_CLOUDEDITION_amd64fre_en-us.iso (5.7 GB)
```
[image]

pbatard, 2023-12-08:
Thanks for the report.

We indeed don't have it [in the code](https://github.com/pbatard/rufus/blob/master/src/stdfn.c#L221-L309).

It looks like we either missed that entry from https://learn.microsoft.com/en-us/windows/win32/api/sysinfoapi/nf-sysinfoapi-getproductinfo or Microsoft added it recently.

I'll try to get this fixed for the next release.

stdin82, 2024-01-09:
> It looks like we either missed that entry from https://learn.microsoft.com/en-us/windows/win32/api/sysinfoapi/nf-sysinfoapi-getproductinfo or Microsoft added it recently.

That list is missing a fair amount of editions

WindowsAddict, 2024-01-09:
@pbatard You can get the latest list of editions with SKU value from `winnt.h` file. 
Install [latest ADK](https://www.microsoft.com/en-us/software-download/windowsinsiderpreviewadk) and find that file here `"C:\Program Files (x86)\Windows Kits\10\Include\10.0.26020.0\um\winnt.h"`

closed 2024-01-10 by pbatard