pbatard/rufus #2381: Rufus Error During Extraction of Image File

ber999, 2023-12-10:
﻿Rufus Error During Extraction of Image File.
Image file is ubuntu-23.10.1-desktop-amd64.iso.
Stick capacity is 8 GB. 
[image]

pbatard, 2023-12-10:
Log please. Or I will close the issue.

ber999, 2023-12-10:
Rufus x64 v4.3.2090
Windows version: Windows 11 Home x64 (Build 22631.2792)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
0 devices found
Found USB 2.0 device 'Ut165 USB2FlashStorage USB Device' (1307:0165)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 983, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0198155C
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 7.5 GB (8085569536 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Reported Grub version: 2.12~rc1
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 23.10.1 amd64'
  Size: 4.8 GB (Projected)
  Uses: GRUB2 (2.12-rc1)
  Uses: EFI
  Note: This ISO uses symbolic links, which may not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: ubuntu-23.10.1-desktop-amd64.iso (4.8 GB)
Will reuse 'core.img' from 'C:\Users\dreye\AppData\Local\Rufus\grub-2.12-rc1\' for Grub 2.x installation

Format operation started
Requesting disk access...
Will use 'J:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Grub 2.0 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Write error [0x0000045D] (FormatMessage error code 0x00003AFC)
Retrying in 5 seconds...
Write error [0x0000045D] (No Windows Error String)
Retrying in 5 seconds...
Write error [0x0000045D] (No Windows Error String)
Retrying in 5 seconds...
Write error [0x0000045D] (No Windows Error String)
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 7.5 GB)
Waiting for logical drive to reappear...
Formatting to FAT32 (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive1 for exclusive write access
Writing Master Boot Record...
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Grub 2.0 MBR
Writing Grub 2.0 SBR (from download) ✓
Found volume \\?\Volume{c30352fd-974d-11ee-a472-c0b6f99409de}\
[... 819 lines cut ...]
Extracting: J:\pool\main\g\gcc-13\libhwasan0_13.2.0-4ubuntu3_amd64.deb (1.2 MB)
Extracting: J:\pool\main\g\gcc-13\libitm1_13.2.0-4ubuntu3_amd64.deb (28.6 KB)
Extracting: J:\pool\main\g\gcc-13\liblsan0_13.2.0-4ubuntu3_amd64.deb (1.1 MB)
Extracting: J:\pool\main\g\gcc-13\libquadmath0_13.2.0-4ubuntu3_amd64.deb (149.3 KB)
Extracting: J:\pool\main\g\gcc-13\libstdc++6_13.2.0-4ubuntu3_i386.deb (821.0 KB)
Extracting: J:\pool\main\g\gcc-13\libstdc++-13-dev_13.2.0-4ubuntu3_amd64.deb (2.2 MB)
Extracting: J:\pool\main\g\gcc-13\libtsan2_13.2.0-4ubuntu3_amd64.deb (2.5 MB)
Extracting: J:\pool\main\g\gcc-13\libubsan1_13.2.0-4ubuntu3_amd64.deb (1.1 MB)
Extracting: J:\pool\main\g\gcc-defaults\gcc_13.2.0-1ubuntu1_amd64.deb (5 KB)
  Unable to create file: [0x0000045D] (No Windows Error String)
Error: Can't write 'J:\md5sum.txt'

Found USB 2.0 device 'Ut165 USB2FlashStorage USB Device' (1307:0165)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 983, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0204A059
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 7.5 GB (8085569536 bytes)
  Start Sector: 2048, Boot: Yes
Saved C:\Users\dreye\rufus_Fehler_beim_Entpacken.log

pbatard, 2023-12-10:
All the `0x0000045D` errors I've seen, and you are getting quite a few, appear to have been caused by defective media.

Please run a bad blocks check against your USB media, or use a different one (which are 2 elements that you were advised to do in the check list you deleted when you opened the issue).

ber999, 2023-12-10:
Thank you for the information.
I have used another stick and the creation of the system worked properly now.

closed 2023-12-10 by ber999