pbatard/rufus #2384: How to Install Windows 11 in S Mode

sanjeet007, 2023-12-15:
You guys added one feature in 4.3 to Windows install in S Mode but I don't find any settings or way to do it any guide 😔

pbatard, 2023-12-15:
And the reason why it is behind an expert mode is because **you are supposed to know what you are doing**, because it can be exceedingly difficult to get a computer our of S Mode, and **ESPECIALLY** you will **NOT** get out of S-Mode by wiping the computer and reinstalling Windows.

So, if you do try S Mode, please dont come asking how to get your computer out of it. If you try an expert feature, you are 100% on your own.

closed 2023-12-15 by pbatard

sanjeet007, 2023-12-15:
Sorry but I did not understand how to do it like while creating bootable usb what to do ?

Please can you tell me steps to follow?

Thank you!

pineapple63, 2023-12-15:
> Sorry but I did not understand how to do it like while creating bootable usb what to do ?
> 
> Please can you tell me steps to follow?
> 
> Thank you!

If you are ABSOLUTELY 100% SURE you want to enable S mode, here are instructions for using Rufus

Note: if the device was previously in S mode and has been switched out of it, these instructions will NOT re-enable S mode (or if that isn’t true, then I don’t know what went wrong when I experimented with putting a cheap Mini PC (that I don’t care about) into S mode (it didn’t go into S mode))

After opening Rufus, Press CTRL + ALT + E, you should see a message on the progress bar that expert mode is now enabled

Select an ISO as normal, and press start, the Windows User Experience dialogue will come up (assuming it is a Windows 10/11 ISO you selected), and at the bottom is the one for S mode (this will NOT appear unless expert mode is enabled)

sanjeet007, 2023-12-15:
Thank you for telling but you were right it did not go in to Windows 11 Home S Mode it's installed normal windows 11 totally time waste 😔 @pbatard what's wrong?