pbatard/rufus #2385: Installing Windows 11 Home S Mode Not Working

sanjeet007, 2023-12-15:
Unfortunately i tried to many steps to achieve this then I found this easy method using Rufus.

But not working.

I used windows 11 Home iso and enabled export mode and select s mode option and create bootable usb and normally installed but what i found my windows not in S Mode.

Why? What's gone wrong here?

Thank you!

pbatard, 2023-12-15:
Expert features are not ones I can support. As I stated in the issue you commented on earlier, if you try an expert feature of Rufus that sits behind a wall, you are 100% on your own.

The feature is hidden because:
1. It is only meant to be used **if** you know what you are doing.
2. It is not guaranteed to work.
3. It will never ever be officially supported, which means that I will not help you even if you open an issue in the issue tracker.

closed 2023-12-15 by pbatard

sanjeet007, 2023-12-16:
Oh that's means it's useless 😞 so remove it from your latest release the point you mentioned it's co fusing it's not working in my any device.

But thank you for confirming ☺️

pbatard, 2023-12-16:
It worked fine for mme when I tested it. And I would assume it should work for people who know what they are doing. So, and let me be absolutely clear about it: If this feature doesn't work for you, then it's not for you.

sanjeet007, 2023-12-17:
This is not the way to guide people 😔 you haven't clear about your feature you said enable export mode i did i get a mode option i created bootable usb but after installation I get   normal windows.

Tell me what exactly you did different from me so you get it.

I am youtuber if it works for me I will create video it will be for your side.

I hope you understand!

Thank you!

sabaasa, 2023-12-24:
why would anyone want to force S mode on their PC's