pbatard/rufus #2388: Feature Request: Add Option to select that Win 10/11 doesn't install Drivers automatically after Installation

ONE11111111ONE, 2023-12-22:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Good Morning,

i have a good idea for a feature request which seems not be hard to implement. 
Like for the moment you have these 4 Options when creating a bootable e.g. Win 10.
Even it's of course not the main part of this tool as i understand, it would be awesome for many many people if there would be an option to NOT let windows 10..or 11, install drivers automatically when connected to the internet after freshly installing the OS, since it is messing up many things many times with drivers from manufacturer. 
if there would be more actual additional drivers which the OS can recognise, it would make sense to let them install via manual windows update search.

What do you think about it?

All the best. 

Log
---
```
no log needed
```

ONE11111111ONE, 2023-12-22:
BTW: i am using your tool since many many years and it helped me A LOT. i can make one feature request after the other, if you want to go that way, i am not able to program such stuff on my own, yet i would be able with wiki of someone to manage the mentioned topic above. since it is not everyones business and ppl need help and are comfy AND don't want to mess up things afterwards. I guess thats why you started to add these option stuff, which are wonderful enhancements in my opinion.

pbatard, 2023-12-22:
Please see https://github.com/pbatard/rufus/wiki/FAQ#user-content-Could_you_add_support_for_removing_ltWindows_option_Xgt_in_the_Windows_User_Experience_dialog as it is relevant to what Rufus is designed for when it comes to Windows customization.

I'm afraid that what you request falls outside of the design goal I have for Rufus, so I am not planning to add what you suggest.

Please also understand that, if I was starting to add all of the Windows customization people want to see Rufus apply (because why should I grant your request and not theirs?), the Windows User Experience would literally have more than 50 different entries and therefore become an eternally growing unnavigable mess.

If you don't want drivers to automatically install, my advice is to disconnect from the network, disbale the Windows Update service, then connect again.

closed 2023-12-22 by pbatard

ONE11111111ONE, 2023-12-22:
i accept and understand your goal of rufus. 
your solution is of course a known and logical option, yet i will somehow built it into the written image myself.
maybe someone is out there already having a script for it, i need to find out. 
thanks for your time. 
keep up the good work!