pbatard/rufus #2393: Rufus doesn't reformat NTFS USB drive

jhc019, 2023-12-28:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
Hi,
I am not a developper but a simple "advanced" user. WHen trying to create a bootable USB with Rufus U run into an interesting issue which you probably know about: I have 64 Gb USB drive checked for errors, fine. I need to create a MBR bootable Win7. I choose NTFS format as Win7 is NTFS and start Rufus. It erased the partition present on the USB and stop there as there is no disl anymore it says. When checked with Disk manager in Windows 10 the partition on USB is effectively became RAW. I tried with another USB drive, the same. I reformat USB dirve with Windows, as NTFS. Redo the Rufus - the same, USB drive is reformated as RAW, so Rufus can't do any action. I format the drive by Windows as exFAT and restart Rufus. And now it reformat the drive and install the Win7 as NTFS.
So basically if my USB drive is already NTFS Rufus erases partition and male it RAW and if it is exFAT Rufus works fine. Well, this is misleading for the most of simple users whoi will believe that their USB drive is damaged. I think it would be nice to modify the code to allow Rufus to proceed whatever initial partition on USB drive could be.

For what it worth, and thak you very much for having created and maintaining this wonderful tool.
Log
---
```
<FULL LOG>
```

pbatard, 2023-12-28:
I need to see your log. I simply cannot help you without the log.

jhc019, 2023-12-28:
Thank you for your reply. I understand but I don't need help: as I said I resolved the issue by putting my USB drive as exFAT formatting, and then Rufus worked fine. My point was to attract ytour attention to the issue where Rufus erase the NTFS partition on the USB drive and then stops there this partition having became RAW. It would be nice if Rufus could in fact work with all type of USB partitions including by the way the RAW one. Indeed why it is unable to work with RAW partitions at first place?

pbatard, 2023-12-28:
> My point was to attract ytour attention to the issue where Rufus erase the NTFS partition on the USB drive and then stops there this partition having became RAW.

Which is __NOT__ something I can look into unless you provide a log. I just don't have enough data, because you assume that everybody else but you will experience the same issue, and I can tell you that this is not the case.

As a result, since I don't have enough data to look into the issue you want me to look into, because you still do not want to provide me with a log that shows exactly what you are doing when the partition becomes RAW, I have no choice but to close this issue (and wait for someone who experience a similar issue and will provide a log to look into it).

closed 2023-12-28 by pbatard

jhc019, 2023-12-28:
Ok, no problem, just want you to let you know about this. This is not that I don't want to give you a log but simply that I don't have one: I resolved the issue and closed Rufus without saaving the log. I might try to reproduce the issue if you think it worth it for you however. Let me know