pbatard/rufus #2394: Rufus 4.3 issue with Win11_23H2_English_x64v2.iso from MS

007mesen, 2023-12-29:
So Ive used Rufus 4.3 to make a bootable usb with Win11_23H2_English_x64.iso downloaded in November from Microsoft and no issues.

When trying the December download of Win11_23H2_English_x64v2.iso on Lenovo laptops results in blue screen during boot.

When this occurred I tried again with Win11_23H2_English_x64.iso from November no issues.

So then I went back to Microsoft downloaded Win11_23H2_English_x64v2.iso verified MD5 and again blue screen.

So Something in Win11_23H2_English_x64v2.iso when done using rufus and booting on a Lenovo e590 using latest bios doesn't agree.

pineapple63, 2023-12-29:
If you want help I suggest you provide a log

cisko99za, 2023-12-30:
same issue with aomei backupper (latest 7.3.3 version)
make iso from aomei, write to pendrive with rufus, got blue screen during boot
but if i write my pendrive directly from aomei, it works fine, my laptop can reboot from pendrive
so the issue is rufus when writing the iso

pbatard, 2023-12-30:
Can't replicate the issue, which, since there is no difference between how Rufus writes a 23H2v2 ISO compared to 23H2, is clearly an issue between Microsoft and the manufacturer of your hardware and **not** Rufus.

Also, nobody seems to care about this issue enough to have bothered to provide a log **as explicitly requested** or indicate where exactly during the boot process the crash occurs (is it before you get to select the installation drive? During 2nd stage boot? Somewhere else?).

Also you should realise that Microsoft is in control of its bootloaders (which are ran **UNMODIFIED** by Rufus), which means that they are free to add whatever new restrictions they want, or introduce bugs that might materialize itself against specific hardware. On the other hand, I strongly suspect that AMOEI use its own bootloaders rather than the Microsoft ones.

As a result of all this, I will close this issue as there is nothing for me to investigate, and I can guarantee that, if you happen to experience a crash, and are running a machine that supports booting from NTFS, you will experience the exact same issue **without using Rufus** if you simply extract the whole content of the ISO onto an NTFS formatted partition (but again, this wil only work on systems that have a UEFI firmware that embeds an NTFS driver, which more and more modern systems do).

closed 2023-12-30 by pbatard

TehLandstander, 2024-01-15:
Had the same issue with Win11_23H2_English_x64v2.iso for a USB stick using Win7 and Rufus 3.22.  Tried reformatting as FAT32, NTFS, looked for firewall/AV issues, security protection of the folder, etc.
After struggling I got it to work by reformatting the stick and UNCHECKING "Quick Format".  It worked after that.