pbatard/rufus #2399: USB drive becomes unformatted after running Rufus

mfitzk, 2024-01-11:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I'm using a WD Elements 1TB external hard drive and an Asus ROG Zephyrus G14 GA402XV-G1, both brand new, with Windows 11. I've been trying to copy the QubesOS ISO file to my hard drive using Rufus following the official Qubes installation guide and Rufus appears to run perfectly, but when the task is finished, my hard drive is no longer recognized in File Explorer and has multiple RAW/unallocated partitions. I can clean the disk just fine using diskpart and start from the beginning (drive recognized by File Explorer and one healthy formatted partition with my assigned letter) without issue, but no matter what I try, it always becomes unformatted. I've tried using different USB ports, using a different hard drive, trying different formats, and using other programs like Etcher, none of which worked.

Log
---
```
<FULL LOG>

Rufus x64 v4.3.2090
Windows version: Windows 11 Home x64 (Build 22621.2861)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 3.0 device 'WD Elements 2621 USB Device' (1058:2621)
Device eliminated because it was detected as a Hard Drive (score 28 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found USB 3.0 device 'WD Elements 2621 USB Device' (1058:2621)
1 device found
Disk type: FIXED, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 121597, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x16A652DB
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 931.5 GB (1000167440384 bytes)
  Start Sector: 2048, Boot: No
Checking for Rufus updates...
Checking release channel...
No new release version found.
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Reported Grub version:
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'QUBES-R4-2-0-X86-64'
  Size: 6.2 GB (Projected)
  Has a >64 chars filename
  Uses: EFI
Using image: Qubes-R4.2.0-x86_64.iso (6.1 GB)
Found USB 3.0 device 'WD Elements 2621 USB Device' (1058:2621)
1 device found
Disk type: FIXED, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 121597, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x16A652DB
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 931.5 GB (1000167440384 bytes)
  Start Sector: 2048, Boot: No
Timeout while retrieving conflicting process list

Format operation started
Requesting disk access...
Will use 'Z:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Requesting logical volume handle...
No logical drive found (unpartitioned?)
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Opened \\.\PhysicalDrive1 for shared write access

Found USB 3.0 device 'WD Elements 2621 USB Device' (1058:2621)
1 device found
No volume information for drive 0x81
Disk type: FIXED, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 121597, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {71186176-C5A6-43CA-94D8-23E0AA332BF9}
Max parts: 248, Start Offset: 32768, Usable = 1000170521088 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'ISO9660'
  Detected File System: ISO9660
  ID: {71186176-C5A6-43CA-94D9-23E0AA332BF9}
  Size: 6.1 GB (6512508928 bytes)
  Start Sector: 64, Attributes: 0x1000000000000005
Partition 2:
  Type: EFI System Partition
  Name: 'Appended2'
  Detected File System: FAT12
  ID: {71186176-C5A6-43CA-94DA-23E0AA332BF9}
  Size: 4.7 MB (4966400 bytes)
  Start Sector: 12719808, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'Gap1'
  Detected File System: (Unrecognized)
  ID: {71186176-C5A6-43CA-94DB-23E0AA332BF9}
  Size: 300 KB (307200 bytes)
  Start Sector: 12729508, Attributes: 0x1000000000000001
```

pbatard, 2024-01-11:
> my hard drive is no longer recognized in File Explorer and has multiple RAW/unallocated partitions.

That is normal.

Windows can only read/mount Windows file system and QubeOS is using Linux file systems, so, of course, Windows does not know how to mount/read these partitions and declares them to be RAW.

In other words, you drive is __not__ unformatted. It is just formatted with a file system that Windows cannot mount.

The question then becomes: Have you tried booting the drive instead of simply assuming that it will not work because Windows cannot see its partitions?

mfitzk, 2024-01-11:
Yes, I'm able to reach the installation menu, but it says that I don't have enough free space.

[image]
[image]
[image]
[image]
[image]

pbatard, 2024-01-11:
Well, that's not a Rufus problem, so you will have to sort it out with the QubeOS support community.

For the record, you need to understand that what you created is an __installation media__ that is supposed to be used once and discarded, to install QubeOS to a target drive.

In other words, if you plan to install QubeOS on your 1 TB drive, then you should use a __different smaller USB drive__ with Rufus to create a boot media that you can then use to install QubeOS on your final drive.

Unless you know what you're doing, you cannot use the same media as installation media and target. You need 2 different drives for that.

Because this is not a Rufus problem, I will close this issue now.

closed 2024-01-11 by pbatard