pbatard/rufus #2400: Help to start with USB key Rufus to reinstall windows 11

Dovarese, 2024-01-15:
Hello,
I need your help.

I need to reinstall Windows 11 and I have created a bootable USB key with Rufus 4.3.

The PC has a UEFI bios and the motherboard is an ASUS ROG TRIX B550-e Gamer; the Windows 11 ISO was recovered using Microsoft's Media creation tool.

When I want to start the PC by booting from the USB key (F8 key) I get to a screen that offers me several choices for booting:

1) Windows Boot manager (M2.1 : Force ; MP600 (500,1 GB)
    This is my PC's system disk

2) UEFI: USB, Partition 1 (30.7GB)
3) UEFI: USB, Partition 2 (30.7GB)
1 and 2 correspond to the USB key used with Rufus

When it was created by Rufus, it was patented.

A primary partition (H) of 28.65 GB in NTFS and a partition of 1.00 MB in FAT12.

Information provided by AOMEI Partition Assistant Professional Edition

Can you tell me if my key is OK and which partition I need to select to start installing Windows 11?
Partition 1 or Partition 2

Below is the Rufus configuration when creating the USB key

Thank you very much for your future help and your Rufus software.

[image]

pbatard, 2024-01-16:
Any of the 2 UEFI partitions should do. There really isn't any difference: either it'll boot to the Windows installer, and you're good, or it won't. But the end result will be __exactly the same__ regardless of the UEFI partition you choose.

closed 2024-01-16 by pbatard

Dovarese, 2024-01-16:
OK and thank you
Have a nice day