pbatard/rufus #2404: Alpha warning when upgraded from Microsoft Store

Corsair-63, 2024-01-17:
Previous version installed from Microsoft store﻿, today gone to it and checked for updates.

found latest version, installed.
when started the program a splash window appears showing the following:
[image]

program layout as follows:

[image]

log for starting program as per below:
```
Rufus x64 v4.4.2103 (Alpha) (AppStore version)
Windows version: Windows 11 Pro x64 (Build 22631.3007)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0C0A (es-ES)
Will use default UI locale 0x0C0A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Notice: Windows VDS is unavailable
```

AFAIK I have not the option to update to Alpha/Beta versions, is there any difference with latest from your web?

also happens in my tablet with windows 11 Pro ARM64, same build number as above .

not a big deal but a cosmetic bug, really happy with the performance of the app either x64 either ARM64 in windows 11.

pbatard, 2024-01-17:
Yup, I'm working on getting this fixed but it needs to go through the App Store approval, and this may take while...

pbatard, 2024-01-18:
Should be fixed now. Make sure you update the app.

closed 2024-01-18 by pbatard

Corsair-63, 2024-01-18:
perfect now not showing the spalsh windows and version displayed 4.4.2103