pbatard/rufus #2405: Preserve file timestamp when copying from ISO image

mcgitty, 2024-01-18:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [n/a] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [n/a] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [n/a] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [n/a] I also tried one or more of the following:
  - [n/a] Using a different USB drive.
  - [n/a] Plugging the USB into a different port.
  - [n/a] Running Rufus on a different computer.
- [n/a] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When Rufus copies files from the ISO image onto the USB drive, the current time is used. It would be a great enhancement if this copy also preserves the file timestamp from the ISO image. Perhaps the reason to use CopyFileA() instead of CopyFile() and other CopyFile???() variants should be reassessed. 

Log
---
N/A

pbatard, 2024-01-18:
Already available and described in the FAQ: https://github.com/pbatard/rufus/wiki/FAQ#user-content-Power_keysCheat_modes

<kbd>Alt</kbd>-<kbd>T</kbd>.

closed 2024-01-18 by pbatard

mcgitty, 2024-01-18:
Power keys/Cheat modes
Note: If a cheat mode has a star (*), it means it gets preserved between runs.
...
Alt-T* (v2.3 or later) - Preserve timestamps when extracting ISO content.