pbatard/rufus #2406: Add Iso download support again for old windows 10 version

raunaksingh765, 2024-01-19:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

raunaksingh765, 2024-01-19:
[rufus.log](https://github.com/pbatard/rufus/files/13986886/rufus.log)

JonnyTech, 2024-01-19:
Previous versions of Windows are no longer available from Microsoft's download servers therefore Rufus cannot provide them - only current builds can now be downloaded.

raunaksingh765, 2024-01-19:
> Previous versions of Windows are no longer available from Microsoft's download servers therefore Rufus cannot provide them - only current builds can now be downloaded.

anyother way to download old versionn of windows again

pbatard, 2024-01-19:
> anyother way to download old versionn of windows again

Cannot LEGALLY be done, as the content is __copyrighted__ by Microsoft and can only be distributed by them.

If I were to host a repository of old Windows 10 ISOs, I would be sued and the Rufus project would be shut down.

Therefore, what you request won't happen and I will close this request.

closed 2024-01-19 by pbatard