pbatard/rufus #2411: windows11 formatting error

fachocn, 2024-01-23:
Hello, I have tried several versions and they all display "formatting error". How can I solve this problem? thanks

Rufus x64 v4.4.2103 (Portable)
Windows version: Windows 11 Home China x64 (Build 22621.3007)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0804 (zh-CN)
Will use default UI locale 0x0804
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'zh-CN'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'Mass Storage Device USB Device' (14CD:125D)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 536 GB, Sector size: 512 bytes
Cylinders: 65270, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {3B5658BC-E06E-404D-9379-B57D25E55B6E}
Max parts: 128, Start Offset: 17408, Usable = 536871926272 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'EFI System Partition'
  Detected File System: (Unrecognized)
  ID: {00C1EBEB-FA98-4AD8-B569-CABA014A9478}
  Size: 260 MB (272629760 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft Reserved Partition'
  Detected File System: (Unrecognized)
  ID: {B4358326-05BF-4237-9EF7-BCC43E1E71C4}
  Size: 128 MB (134217728 bytes)
  Start Sector: 534528, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: (Unrecognized)
  ID: {F45C1E90-5337-44DE-8099-88DD20A2D613}
  Size: 499.6 GB (536464044032 bytes)
  Start Sector: 796672, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CPBA_A64FRE_ZH-CN_DV9'
  Detected: Windows 11 ISO (Build 22621.2428)
  Size: 6.6 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (UEFI only)
  Uses: Install.wim (version 0.13.1)
Using image: SW_DVD9_Win_Pro_11_23H2_Arm64_ChnSimp_Pro_Ent_EDU_N_MLF_X23-59518.ISO (6.7 GB)
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM0/sources/install.wim:[0] (API)
Extracting: C:\Users\facho\AppData\Local\Temp\Ruf2C8F.tmp (From [1].xml)
Closing: \\.\CDROM0/sources/install.wim
Will use 'Windows 11 专业版' (Build: 22621, Index 2) for Windows To Go
Selected Windows User Experience options:
• Bypass online account requirement
• Set internal drives offline

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive2 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating EFI System Partition (offset: 1048576, size: 260 MB)
● Creating Microsoft Reserved Partition (offset: 273678336, size: 128 MB)
● Creating Main Data Partition (offset: 407896064, size: 499.6 GB)
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
[0x00000002] The system cannot find the file specified. - Retrying...
[0x0000012A] Too many posts were made to a semaphore. - Retrying...
[0x0000012A] Too many posts were made to a semaphore. - Retrying...
[0x0000012A] Too many posts were made to a semaphore. - Retrying...
Format error: Undetermined error while formatting.
Re-mounted volume as F: after error

Found USB 2.0 device 'Mass Storage Device USB Device' (14CD:125D)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 536 GB, Sector size: 512 bytes
Cylinders: 65270, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {9A14ACA8-3F39-40D9-8F09-222B70285E10}
Max parts: 128, Start Offset: 17408, Usable = 536871926272 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'EFI System Partition'
  Detected File System: (Unrecognized)
  ID: {613C88B7-B82E-4A7D-92D3-5966B23CFA59}
  Size: 260 MB (272629760 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft Reserved Partition'
  Detected File System: (Unrecognized)
  ID: {2CA65C83-BA57-425A-875E-13319F7527EF}
  Size: 128 MB (134217728 bytes)
  Start Sector: 534528, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: (Unrecognized)
  ID: {EA895B93-EF02-4D73-89F4-95AC91F01C38}
  Size: 499.6 GB (536464044032 bytes)
  Start Sector: 796672, Attributes: 0x0000000000000000
[image]

pbatard, 2024-01-23:
Per the checklist:
---------

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.

At least, please run the bad blocks check, as you were requested.

fachocn, 2024-01-24:
@pbatard 
Hello, I conducted a night long bad block detection yesterday and it was reported that there were 287 pieces. I am currently using a TF card, can this situation still be repaired? Also, I bought a Sata solid-state drive case and will try it again tomorrow or the day after the delivery arrives.

Rufus x64 v4.4.2103 (Portable)
Windows version: Windows 11 Home China x64 (Build 22621.3007)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0804 (zh-CN)
Will use default UI locale 0x0804
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'zh-CN'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'Mass Storage Device USB Device' (14CD:125D)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 536 GB, Sector size: 512 bytes
Cylinders: 65270, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {9A14ACA8-3F39-40D9-8F09-222B70285E10}
Max parts: 128, Start Offset: 17408, Usable = 536871926272 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'EFI System Partition'
  Detected File System: (Unrecognized)
  ID: {613C88B7-B82E-4A7D-92D3-5966B23CFA59}
  Size: 260 MB (272629760 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft Reserved Partition'
  Detected File System: (Unrecognized)
  ID: {2CA65C83-BA57-425A-875E-13319F7527EF}
  Size: 128 MB (134217728 bytes)
  Start Sector: 534528, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: (Unrecognized)
  ID: {EA895B93-EF02-4D73-89F4-95AC91F01C38}
  Size: 499.6 GB (536464044032 bytes)
  Start Sector: 796672, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CPBA_A64FRE_ZH-CN_DV9'
  Detected: Windows 11 ISO (Build 22621.2428)
  Size: 6.6 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (UEFI only)
  Uses: Install.wim (version 0.13.1)
Using image: SW_DVD9_Win_Pro_11_23H2_Arm64_ChnSimp_Pro_Ent_EDU_N_MLF_X23-59518.ISO (6.7 GB)
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM0/sources/install.wim:[0] (API)
Extracting: C:\Users\facho\AppData\Local\Temp\Ruf542C.tmp (From [1].xml)
Closing: \\.\CDROM0/sources/install.wim
Will use 'Windows 11 专业版' (Build: 22621, Index 2) for Windows To Go
Selected Windows User Experience options:
• Bypass online account requirement
• Set internal drives offline

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive2 for shared write access
[... 386 lines cut ...]
Cylinders: 65270, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {501E5155-CAD9-4EA7-A761-2BEB3DEED0CA}
Max parts: 128, Start Offset: 17408, Usable = 536871926272 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'EFI System Partition'
  Detected File System: (Unrecognized)
  ID: {87E7EB15-B792-47F3-887C-80A0A23A88B4}
  Size: 260 MB (272629760 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft Reserved Partition'
  Detected File System: (Unrecognized)
  ID: {A67F8CCE-397E-42D4-BF7C-4304BF9049B4}
  Size: 128 MB (134217728 bytes)
  Start Sector: 534528, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: (Unrecognized)
  ID: {8AA86CB0-7C69-43E3-95A7-076F7BDC77B1}
  Size: 499.6 GB (536464044032 bytes)
  Start Sector: 796672, Attributes: 0x0000000000000000

pbatard, 2024-01-24:
> I conducted a night long bad block detection yesterday and it was reported that there were 287 pieces

That __is__ your issue. And Rufus stops reporting bad blocks when it detects more than a specific number.

> can this situation still be repaired?

No. If you value your data and your time, you should consider that your media is irreversibly damaged when a bad blocks test report multiple defective block.

Because bad media is clearly the cause of all your problems, I will close this issue.

closed 2024-01-24 by pbatard