pbatard/rufus #2412: Feature Request: Works with Ventoy (create ISO)

Comettii, 2024-01-24:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Firstly, thank you for your work.
I would like to suggest a new functionality, which would make RUFUS apply tweaks like it already does today with the options to remove TPM limitation, local account, etc., but instead of burn the ISO on removable media, it would give us an option to create an new ISO with these modifications, so we would be able to work with Ventoy for example.
This would open up countless possibilities and a new approach to using RUFUS, after all, over time there could be more registry modifications offered by RUFUS, so that we could customize our ISOs and use them as usual or in this way, saving them in a new ISO file updated.

Log
---
```
<FULL LOG>
```

JonnyTech, 2024-01-24:
See @pbatard's reply at https://superuser.com/questions/1489113/getting-rufus-to-output-to-a-disk-image-file

pbatard, 2024-01-24:
Also see https://github.com/pbatard/rufus/wiki/FAQ#user-content-Do_you_plan_to_add_USB__ISO_creation_support which is the official FAQ answer.

You can easily create an ISO on your own. Or you can create a VHD/VHDX from Rufus, which I believe works with Ventoy. There's no point to add what you suggest.

closed 2024-01-24 by pbatard