pbatard/rufus #2415: English text are misspelled.

Legendarion, 2024-01-28:
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------
I believe the english text are misspelled in the small context menus (if it's that it is called).

The word 'disk' should be 'disc'.

[image]

[image]

[image]

I assume the sentence should be "Use at your own risk" and not 'risks'(?)
[image]

Log
---
```
Rufus x64 v4.4.2103
Windows version: Windows 10 Home x64 (Build 19045.3996)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0414 (nb-NO)
NOTE: No Windows Language Pack is installed for English (English) on this system.
This means that some controls may still be displayed using the system locale.
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
0 devices found
```

pineapple63, 2024-01-28:
Disc is the spelling used for optical media, all other types of storage media use the spelling Disk

Legendarion, 2024-01-28:
OK.

closed 2024-01-28 by Legendarion