pbatard/rufus #2416: Missing "Windows User Experience" when ISO on a Shared Folder

cciRRus, 2024-01-28:
I am using the latest **Rufus 4.4** on a Windows 11 Pro 23H2 virtual machine (hypervisor is VMware Fusion 13.5.0). I enabled "Sharing" so that the VM is able to access the ISO file **Win11_23H2_English_x64.iso** on my Mac "Downloads" folder.

On the Windows 11 VM, I ran Rufus, and selected the file from Z:, which is actually automatically mapped \\vmware-host. The full path of the source ISO is therefore **\\vmware-host\Shared Folders\Downloads\Win11_23H2_English_x64.iso**

When I pressed the Start button, the Windows User Experience window did not appear. Rufus continued to format the USB disk then wrote the files over.

To resolve the issue, I copied the ISO file to the VM Desktop folder. And with Rufus, I selected that ISO on my Desktop. The Windows User Experience window appeared.

pbatard, 2024-01-28:
Similar to what people are experiencing with parallels on Mac.

I would advise you to contact VMWare to tell them that their emulation is not working 100% as it is breaking the Windows WIM API which appears to be unable to open data of the __virtually__ mapped drive. The goal of emulation is to provide a virtual machine that behaves 100% like a real one. If native Windows APIs get broken as a result of the emulation, then clearly the emulation needs to be fixed.

Also, next time, please provide a log, as you were explicitly requested to do, as you would have seen a message there telling you that Rufus was unable to open the WIM file, that it needs to process to be able to show the WUE dialog.

closed 2024-01-28 by pbatard

cciRRus, 2024-01-28:
I am trying to inform whoever is facing the problem, that the solution is to simply copy the file over to the VM disk instead of going over the Shared Folder. As such, I did not share any logs.

pbatard, 2024-01-28:
> As such, I don't see why the other users would have cared about my logs. 

Because if other users encounter a similar issue, the log will tell them if it's the same or not.

Again you __chose__ to disregard something that was asked of you to do, because you assumed that you know better than the developer of the application what's actually needed when reporting an issue. And you are not providing a "solution", but a workaround, so, unless you are contacting VMWare to report this issue, you are not going to help other users in the same situation as yours as much as you believe you do.