pbatard/rufus #2417: Thanks for the test. I probably should have checked the driver from 2.18 before I asked you to run that test.

pgsohail, 2024-01-29:
hello,
hope u are good 
so i need ur help i am currently facing a similar issue i was previously using windows 11 and then i switched to fedora now i want to install windows 10 again in my laptop and i am unable to install it as i am facing these types of issues i am also attaching the scrrenshots below i have bought 16gb of usb and 32gb and a 64gb usb to but all are giving same errors after making a bootable usb 
also i am making my bootable usb from rufus also i have tried ventoy too.
here are the screenshots and error outputs 

LINFO]
UEFT v2.70 CHP, 0x01090200)
LINFO]
HP U99 Ver. 01.09.02
LINFO]
CINFO]
HP HP Book Studio 16 inch G9 Mobile Workstation PC
Secure Boot status: Disabled
LINFO1 LINFO)
Disconnecting potentially blocking drivers
LINFO]
Searching for target partition on boot disk:
PciRoot (0x0) /Pci (0x14,0x0) /USB (0x5,0х0)
LINFO]
Found NTFS target partition:
CINFO]
LINFOJ
PciRoot (0x0)/Pci (0x14,0x0)/USB (0x5,0x0) /HD (1,GPT, CFFA2155-E75D-44C9-92EB-1D7BEFBFA49D.0x800.0x75F6FAB)
LWARNI
Checking if target partition needs the NIFS service
An NIFS driver service is already loaded
LINFO]
Opening target NIPS partition:
LINFO]
Volume label is 'CCCOMA_X64FRE_EN-GB_DV9*
LINFOI LINFO]
This system uses 64-bit x86 UEFI → searching for x64 EFI boot loader
Launching
CFAILI
'ef/\boot\bootx64.efl*
Load failure: [3]
・・・
Unsupported
[image]
[image]
[image]
[image]

pbatard, 2024-01-29:
I will at the very least need the log of how you created the drive in Rufus. The log you posted (from UEFI:NTFS) is not enough.

Without this, I can't help you, and you were explicitly asked to provide that log (as well as told how you could access it) when you opened the issue.

pgsohail, 2024-01-29:
bro can u tell me how to acess the log 
as i dont know please

pgsohail, 2024-01-29:
also i have attached the screenshots too u can have a look

pbatard, 2024-01-29:
It was stated explicitly in the checklist that you saw when you opened the issue, but decided to __delete__:

> - [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
> - [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Also, I would strongly advise to perform the following, since chances are you are using a broken Windows 10 ISO.

Please do yourself a favour, and ditch the ISO you are using. Then either go to https://www.microsoft.com/en-US/software-download/windows10 (if you are doing it from Linux, as if you are accessing the Microsoft site from Windows, it will not let you downoald the ISOs directly) or use Rufus to download a new official Windows 10 ISO:

[image]

Then you will need to perform [this](https://github.com/pbatard/rufus/wiki/FAQ#user-content-How_can_I_validate_that_a_Windows_ISO_is_a_genuine_retail_version) to validate that your ISO is not corrupted: 

Only after you have done that should you try to create a bootable media using that ISO.

pgsohail, 2024-01-29:
thanks bro i will do this and get back to u in some time

pbatard, 2024-01-29:
> also i have attached the screenshots too u can have a look

The screenshots tell me that, for whatever reason, the `bootx64.efi` from your ISO is broken.

It can be 3 things:
1. Your ISO is corrupted.
2. The NTFS driver from your computer (which comes from HP and not from Rufus) is buggy (especially if you only see one option for boot when you plug the USB media created by Rufus, as, if your computer does have a __working__ native NTFS driver, you should see two options).
3. There's an incompatibility between your HP hardware and the Windows bootloader.

Case 1 is something you can easily act upon by making sure your ISO is not corrupted.
Case 2 and 3 are something only HP or Microsoft will be able to resolve.

pgsohail, 2024-01-29:
ok thanks

pgsohail, 2024-01-29:
i have tried it but its the same error on a differnt laptop too 
it shouldnt be the same error on the two differnt laptops also i have used rufus 4.4 x86 version and again i used simple rufus 4.4 version 
the errors are the same on both laptops 
also i have downloaded the iso file from the link u gave its official how can it be broken? 

can u please provide me some urgent solution to this as i have very important task please

pbatard, 2024-01-29:
Again, I need to see your log.

pgsohail, 2024-01-29:
sure let me

pgsohail, 2024-01-29:
Rufus x86 v4.4.2103
Windows version: Windows 11 Home x64 (Build 22631.3085)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {3DDDB754-065D-44EF-A4CC-02FA44C65F4B}
Max parts: 128, Start Offset: 17408, Usable = 31029426688 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {BD528F94-7484-4AD8-90C7-0643EEC054D7}
  Size: 28.9 GB (31027302400 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {A83EE873-4EA5-473C-943C-58DED9AC0274}
  Size: 1 MB (1048576 bytes)
  Start Sector: 60602248, Attributes: 0x9000000000000000
Checking for Rufus updates...
Checking release channel...
Updates: Unable to access version data

pbatard, 2024-01-29:
The log of how you created the media. **NOT** the log after.

Please recreate your media in Rufus exactly as you did, and post __that__ log.

pgsohail, 2024-01-29:
ok

pgsohail, 2024-01-29:
here u gi bro 

Rufus x86 v3.10.1647
Windows version: Windows 10 64-bit (Build 22631.3085)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {3DDDB754-065D-44EF-A4CC-02FA44C65F4B}
Max parts: 128, Start Offset: 17408, Usable = 31029426688 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {BD528F94-7484-4AD8-90C7-0643EEC054D7}
  Size: 28.9 GB (31027302400 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {A83EE873-4EA5-473C-943C-58DED9AC0274}
  Size: 1 MB (1048576 bytes)
  Start Sector: 60602248, Attributes: 0x9000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_EN-GB_DV9'
  Size: 5.7 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_22H2_EnglishInternational_x64v1.iso (5.7 GB)
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {3DDDB754-065D-44EF-A4CC-02FA44C65F4B}
Max parts: 128, Start Offset: 17408, Usable = 31029426688 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {BD528F94-7484-4AD8-90C7-0643EEC054D7}
  Size: 28.9 GB (31027302400 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {A83EE873-4EA5-473C-943C-58DED9AC0274}
  Size: 1 MB (1048576 bytes)
  Start Sector: 60602248, Attributes: 0x9000000000000000
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 28.9 GB)
● Partition 2 (offset: 31028350976, size: 1 MB)
Opened \\.\PhysicalDrive1 for exclusive write access
[... 781 lines cut ...]
Extracting: F:\sources\replacementmanifests\upnpdevicehost-server-replacement.man (2.4 KB)
Extracting: F:\sources\replacementmanifests\upnpssdp-replacement.man (2.3 KB)
Extracting: F:\sources\replacementmanifests\upnpssdp-server-replacement.man (2.3 KB)
Extracting: F:\sources\replacementmanifests\usbmigration-replacement.man (1.6 KB)
Extracting: F:\sources\replacementmanifests\video-tvvideocontrol-replacement.man (914 bytes)
Extracting: F:\sources\replacementmanifests\virtualdiskservice-repl.man (1.7 KB)
Extracting: F:\sources\replacementmanifests\volsnap-repl.man (2.9 KB)
Extracting: F:\sources\replacementmanifests\vssservice-repl.man (4 KB)
Extracting: F:\sources\replacementmanifests\wbiosrvc-replacement.man (1.2 KB)
Extracting: F:\sources\replacementmanifests\wcf-http-activation-replacement.man (850 bytes)
Extracting: F:\sources\replacementmanifests\wcf-nonhttp-activation-replacement.man (860 bytes)
Extracting: F:\sources\replacementmanifests\web-services-for-management-core-replacement.man (13.4 KB)
Extracting: F:\sources\replacementmanifests\webdavredir-clientonly-replacement.man (2.1 KB)
Extracting: F:\sources\replacementmanifests\webdavredir-serveronly-replacement.man (2.1 KB)
Extracting: F:\sources\replacementmanifests\wia-coreservices-replacement.man (969 bytes)
Extracting: F:\sources\replacementmanifests\wicamigrationav-rl.man (9.9 KB)
Extracting: F:\sources\replacementmanifests\wid8-migration.man (2 KB)
Extracting: F:\sources\replacementmanifests\win32k-settings-repl.man (18.3 KB)
Extracting: F:\sources\replacementmanifests\windows-senseclient-service-rep.man (1.3 KB)
Extracting: F:\sources\replacementmanifests\windows.cortana.desktop-repl.man (1.5 KB)
Extracting: F:\sources\replacementmanifests\windows.holographic.displaythrottling-replacement.man (2.6 KB)
Extracting: F:\sources\replacementmanifests\windows.media-migration-replacement.man (993 bytes)
Extracting: F:\sources\replacementmanifests\windows.sharedpc.accountmanager-repl.man (1.4 KB)
Extracting: F:\sources\replacementmanifests\windows.ui.shell-repl.man (2 KB)
Extracting: F:\sources\replacementmanifests\windowspushnotifications-platfor

pgsohail, 2024-01-29:
Phase duration (Orphan reconnection): 4.14 milliseconds.
 Phase duration (Orphan recovery to lost and found): 0.03 milliseconds.
  0 reparse records processed.
  0 reparse records processed.
 Phase duration (Reparse point and Object ID verification): 16.17 milliseconds.

Stage 3: Examining security descriptors ...
Security descriptor verification completed.
 Phase duration (Security descriptor verification): 14.48 milliseconds.
  96 data files processed.
 Phase duration (Data attribute verification): 5.56 milliseconds.

Windows has scanned the file system and found no problems.
No further action is required.

  30300101 KB total disk space.
   6009980 KB in 912 files.
       344 KB in 98 indexes.
     44093 KB in use by the system.
     41472 KB occupied by the log file.
  24245684 KB available on disk.
      4096 bytes in each allocation unit.
   7575025 total allocation units on disk.
   6061421 allocation units available on disk.
Total duration: 239.71 milliseconds (239 ms).
NTFS Fixup completed.

Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {221D7D36-2834-4400-89DC-53770522FE94}
Max parts: 128, Start Offset: 17408, Usable = 31029426688 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {56588CBA-40F2-4A37-8E1D-D1217C74382A}
  Size: 28.9 GB (31027304448 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {753CCED9-C79C-47C8-BBC0-00D10730D83E}
  Size: 512 KB (524288 bytes)
  Start Sector: 60602252, Attributes: 0x0000000000000000

pgsohail, 2024-01-29:
[image]

Got this error this time

pgsohail, 2024-01-29:
On the other laptop same error 
[image]

This is with version 3.9 rufus

pbatard, 2024-01-29:
> `Rufus x86 v3.10.1647`

Dude!

Why the heck are you using a __very old__ version of Rufus?
Please create your media with the latest  (currently v4.4), and only report if you saw and issue then.

pgsohail, 2024-01-29:
I am having issues with 4.4 too

pgsohail, 2024-01-29:
I saw in issues someone said that 3.10 and 3.9 are better versions that doesn't cause issues

pbatard, 2024-01-29:
Then, especially as you report that you encounter errors regardless of the utility you tried (Ventoy, Microsoft Media Creation Tool) this is an incompatibility between your hardware and Windows, which you will have to solve with your hardware manufacturer or Microsoft.

There's no way all of Ventoy, Rufus and MCT will all fail to boot unless:
- You misconfigured your hardware in the BIOS/UEFI settings (but we can't help you there, because settings are manufacturer dependent and this is not HP or whatever support)
- Your hardware and the latest Windows bootloaders are incompatible.

I know you'd like someone to offer a magic "Oh, you just need to do this, and then that" solution here, but if 3 utilities, that use completely different methods of crafting a boot media, including the official OS' manufacturer one (MCT), lead to media that does not boot on your system, this is an issue that you need to see with your hardware manufacturer.

pgsohail, 2024-01-29:
Yeah sure thanks bro 
I will look into this

closed 2024-01-31 by pgsohail