pbatard/rufus #2420: rufus 4.4 do not recognize my usb ext drive...

p060477, 2024-02-01:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
Hi,
i'm trying to do an usb bootable with an  .iso file
but rufus do not recognize my usb ext hd,
is a western digital my passport essential 
mod. wd1600me-00 , of 160gb
could you pls help me...?
thxs so much again
cheers

Log
---
```
<FULL LOG>
```

pbatard, 2024-02-01:
You do know I am very tempted to just close this issue, as you were explicitly warned when you opened it, because you __CLEARLY__ did not bother to follow the checklist, and just ticked the checkboxes without performing the actions.

If you did, then you would have found the relevant FAQ entry, or, even if you didn't, and you provided the log as requested, the log would also have provided an answer to your questions.

So, instead of pointing you directly to the answer to your question, I will leave you perform what you were requested to do, because if you do, you will find the answer yourself, either in the FAQ or in the log.

closed 2024-02-01 by pbatard

p060477, 2024-02-01:
this is the log...and i've seen it...but i really do not understand why the device was eliminated by rufus...in it there is also written to email the author...and i've also promtly emailed the author about...:

Rufus x64 v4.4.2103 (Portable)
Windows version: Windows 10 Home x64 (Build 19045.3930)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0410 (it-IT)
Will use default UI locale 0x0410
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'it-IT'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'WD 1600BEV External USB Device' (1058:0704)
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'Generic- Compact Flash USB Device' (058F:6364)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- SD/MMC USB Device' (058F:6364)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- MS/MS-Pro USB Device' (058F:6364)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- SM/xD-Picture USB Device' (058F:6364)
Device eliminated because it appears to contain no media
0 devices found

pbatard, 2024-02-01:
Hint:

```
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
```

pbatard, 2024-02-01:
Also pay attention to the: "If this device is not a Hard Drive, please e-mail the author of this application". Your wd1600me __is__ an external hard drive, so there was no need to e-mail the developer.

p060477, 2024-02-01:
first thxs for yr kind attention:
" You can enable the listing of Hard Drives under 'advanced drive properties'"....
how to...??

p060477, 2024-02-01:
[image]

pbatard, 2024-02-01:
How about you spend a little time looking at the FAQ to see if it explains this further (which it does)?

p060477, 2024-02-01:
have you seen my photo attached above...i've read and done...but ..still not detected...why...??

pbatard, 2024-02-01:
Again, your very question is answered in the FAQ. With a picture and everything.

_"Give a man a fish, etc."_

p060477, 2024-02-01:
here i am...the explanation in the faq is very easy and i've done litterally...:
https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_my_USB_drive_is_not_detected

[image]

but as you can easly see the hhd is still NOT detected...
why...??

pbatard, 2024-02-01:
Can you please post your full log again, with the "List USB Hard Drive" option checked (because the log you posted earlier __definitely__ did not have that option checked). You may also want to press <kbd>Alt</kbd>-<kbd>.</kbd> in Rufus before posting that log, so that I get extra info.

pbatard, 2024-02-01:
Considering that you didn't reply, I'm going to assume that you were running the Rufus exe from the drive or did something else that was pretty obvious to pick up when looking at the log (such as leaving your drive unplugged), because there's just no way you could have `Found USB 2.0 device 'WD 1600BEV External USB Device' (1058:0704)` earlier in your log and not have Rufus properly show it once you enable the list HDD option.

p060477, 2024-02-01:
my screenshoot attached posts ago, 2 hours ago...,
 is as much clear as at least yr faqs...:
there is the flag on:
"elenco unità disco"
and at the bottom left is clearly written:
"rilevati 0 dispositivi"

pbatard, 2024-02-01:
Again, I asked you to post your log when you see that screenshot, preferably with <kbd>Alt</kbd>-<kbd>.</kbd>  enabled, because it will be different from when you had the `Device eliminated because it was detected as a Hard Drive (score 13 > 0)` line.

Post your full log again, as requested. The screenshot does not tell us anything.

p060477, 2024-02-01:
i already post a log
i already read the faqs
i already attached here, hours ago..the screenshoot
i do not want to waste yr and also my precious time so i go ahed and put in a flash usb pendrive to solve the issue
finally just i'm using rufus to build a usb bootable with win 10.iso in
which part. should have i to choose? Mbr or Gpt?
and which destination:
Uefi
or dual Uefi/Bios?
i ask it cause this web page make me rise the qurestion:
https://www.hirensbootcd.org/usb-booting/
thxs for yr attention
cheers

pbatard, 2024-02-01:
> i already post a log

A log that does **NOT** match the one you would get from Rufus in the screenshot you posted, and you know it.

> i already attached here, hours ago..the screenshoot

And I have towd you, twice already, that this wasn't enough, and that I needed to see the log that corresponds to that screenshot if you want to put forward the possibility that Rufus may still not detect your driven even with the _List USB Hard Drives_ option selected.

> i do not want to waste yr and also my precious time 

Well, it is clear to me that you are just unable or, more realistically, unwilling to follow directions, which is indeed a waste of time, so I will not help you any further.

pbatard, 2024-02-01:
What part of "If you are being __explicitly TOLD__ to do something, but clearly didn't do it, and then are being __taught__ that there is value in not ignoring stuff that you feel you are above doing, so that you don't waste everybody's time, including yours"_ is not educative?

I guess they should add an addendum to the proverb that it will only work if someone is not actively reluctant to being taught how to fish. Which is what you have demonstrated over and over from the very start. Someone here has been trying to teach you how to fish but you have been rejecting that lesson, because, as it happens, being taught something doesn't mean a zero-effort operation.

p060477, 2024-02-01:
**feel free to think it**
thanks so much for all your efforts and precious patience spent
best regards