pbatard/rufus #2423: Rufus Suggestion: Pausing

WavexDK, 2024-02-03:
So it’s about pausing fx. the ”Windows To Go” option, where you could temporarily ”let the usb drive breathe”, fx. It could not pause immediately but pause after the current process is done, like its currently unpacking ”xyz.dll” and when that is done unpacking, it pauses until you un-pause it.
This would primarily help with thermal throttling, as a lot of usb drives, including mine, decrease their speed after it reaches a certain temperature.

pbatard, 2024-02-03:
Too specific.  Too few people would ever benefit from it, and as hardware improves, this is expected not to be that needed. Oh, and it wouldn't be that simple to add as it is a UI change, and any change to the UI requires a lot more effort that people realize.
See also: https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_cant_you_just_add_the_feature_I_want

I can only suggest that you buy better hardware instead of letting manufacturers getting away with this crap. Storage media should __NOT__ see degraded performance just because the user happens to write at high speed for too long, and software should never ever be required to work around this.

&rarr; Rejected.

closed 2024-02-03 by pbatard

WavexDK, 2024-02-03:
That's totally fine, though you did not have to be that big of a dick about it.

pbatard, 2024-02-03:
I'm just providing a comprehensive list of the reasons why I am rejecting the suggestion. From your comment, I am pretty confident that, had I not done that, you would have accused me of being "a dick" for simply rejecting it without justifying my reasoning for doing so. And I am just stating the truth: Manufacturers are selling subpar hardware, and it's the consumer's responsibility to either reject it, so that they get better product, or access the consequences, such as lower perf/heat issues.