pbatard/rufus #2424: Win 10 - 22H2 is 20H1?

QuePast, 2024-02-04:
Hey, I used Download in rufus, you can check settings in image. But when I mount it and check it it shows 19041 version, which is 20H1 instead. Can you confirm that this is not user error, I am getting confused thanks a lot.
[image]

[image]

QuePast, 2024-02-04:
I guess it works like this.

closed 2024-02-04 by QuePast

JonnyTech, 2024-02-04:
Try [/Get-ImageInfo](https://learn.microsoft.com/en-us/windows-hardware/manufacture/desktop/dism-image-management-command-line-options-s14?view=windows-11#get-imageinfo)

pbatard, 2024-02-04:
The problem is that, when Microsoft generate the WIM files, __they don't bother PROPERLY__ update the version in it (despite the fact that it mostly comes from XML metadata that you'd think that they should long have figured out how to update). Instead, they just seem to copy/paste whatever old XML data they had lying around, and update a few things here and there, while typically not bothering to update the version field.

So, when you look at a wim, you usually get incorrect information about the version, and the __ONLY__ way to actually find the version is to go through a full install of the image and run `winver`.

This is also a major annoyance to me when I update the downloads list, because I need to go through a full Windows To Go install in order to get the real version of the content, so that I can actually show it to the user...

thecatontheceiling, 2024-02-11:
Windows 10 after the release of 2004 (or 20H1) has not gone through any changes in it's codebase and all of the later "feature updates" have technically been cumulative updates. Windows 10 22H2 is technically still 2004, but with cumulative updates that update Windows branding so that it says 22H2 instead. This is normal and can be ignored, as you will experience a similar issue when checking Windows 11 23H2 ISO's, as those ones will say "22H2" when trying to check the wim, as 23H2 is just a extremely small cumulative update for 22H2 and therefore, it is actually still 22H2 under the hood.

thecatontheceiling, 2024-02-11:
also @pbatard you can also check image like this so you don't have to install it:

```
dism /list-image /imagefile:"G:\sources\install.wim" /Index:1 | find /i "Enablement-Package"

```

pbatard, 2024-02-11:
@thecatontheceiling, we're talking about the moniker that is commonly used to describe the release date of the ISO, which Microsoft officially uses. In other words, we want to know the actual version, including build number that the release labelled `YYMM` by Microsoft will produce once installed, and without any subsequent update applied.

I think people here are well aware of the nature of the feature updates, but that doesn't detract from the fact that, because Microsoft has made it a habit of releasing multiple versions of the same `YYMM` (for instance, there have been `Win11_22H2_EnglishInternational_x64.iso`, then `Win11_22H2_EnglishInternational_x64v1.iso` then (after BlackLotus was discovered_ `Win11_22H2_EnglishInternational_x64v2.iso`) released a couple months after the other. For this reason, it is not enough to label an ISO with `YYMM` alone, but we also need to find the actual build number that `winver` will display once you install that specific release off Windows.

And I'm also afraid that your commandline does not produce the data we are after. For instance, for the `install.wim` of `Win11_22H2_EnglishInternational_x64v2.iso` (which reports Build `22631.2861` once installed), your command will produce:
```
C:\tmp>dism /list-image /imagefile:"C:\tmp\install.wim" /Index:1 | find /i "Enablement-Package"
\Windows\servicing\Packages\Microsoft-Windows-23H2Enablement-Package~31bf3856ad364e35~amd64~~10.0.22621.2355.cat
\Windows\servicing\Packages\Microsoft-Windows-23H2Enablement-Package~31bf3856ad364e35~amd64~~10.0.22621.2355.mum
\Windows\System32\CatRoot\{F750E6C3-38EE-11D1-85E5-00C04FC295EE}\Microsoft-Windows-23H2Enablement-Package~31bf3856ad364e35~amd64~~10.0.22621.2355.cat
```
As a matter of fact, the version reported by checking the Enablement-Package as you propose is even worse than the (still inaccurate) version you can get from the WIM XML, which is `22621.2861`. And if you tested on the latest 22H2 ISO, the version you got is definitely not the one we want to see, which should be 19045.2965`.

So, while I appreciate the effort, I'm afraid you're still not helping us answer the simple question which was at the core of this issue: How can one get the __exact__ version that Windows will report once installed, from an `install.wim`, without actually installing the image? And, whereas we've tried many things (like extracting some of the kernel DLLs to poke at their version numbers and whatnot), I'm still not seeing a proper answer to that.

For reference, the build numbers we want to see extracted from the latest Windows 11 and Windows 10 ISOs are the ones you will find [here](https://github.com/pbatard/Fido/blob/master/Fido.ps1#L141) and [here](https://github.com/pbatard/Fido/blob/master/Fido.ps1#L149), so if you can actually find a command that will produces those from the `Win11_23H2_English_x64v2.iso` and `Win10_22H2_English_x64v1.iso`, then you'll have found something that is of interest to us.

thecatontheceiling, 2024-02-11:
ah, I am sorry, I misread the issue and I wanted to try to help

pbatard, 2024-02-11:
No worries. I appreciate your willingness to help on the matter and I'm pretty sure there's some data, somewhere in the `.wim`, that does show the build version we are after, so you're probably on the right track in trying to locate some specific files.