pbatard/rufus #2425: Rufus simply closes when clicked on 'Start' button

abhi-deshpande, 2024-02-05:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
I am trying to create a bootable USB drive with Win11_23H2_EnglishInternational_x64v2.iso. The file has been downloaded from Microsoft's official website and I have successfully used it with Ventoy for one of my friend's computer. The image does not seem tampered anyway.

When I select this image and select GPT - UEFI (non CSM) options (BTW they are the default when image is loaded), and click on 'Start', Rufus closes directly.
Interestingly, When I change it to MBR - BIOS (or UEFI-CSM), Rufus correctly prompts me the warning about USB drive formatting. Does the image has some limitation being a UDF image or is it a bug in Rufus ?

I have used DebugView 4.90.

Log
---
```
00000001	0.00000000	[8832] *** Rufus init *** 	
00000002	0.00783340	[8832] Cur dir: 'C:\Windows\System32\' 	
00000003	0.00785910	[8832] App dir: 'C:\Program Files\WindowsApps\19453.net.Rufus_4.4.2104.0_x64__y8nh7bq2a8dtt\rufus\' 	
00000004	0.00789660	[8832] Sys dir: 'C:\Windows\system32' 	
00000005	0.00790800	[8832] Usr dir: 'C:\Users\abhishek' 	
00000006	0.00793550	[8832] Dat dir: 'C:\Users\abhishek\AppData\Local' 	
00000007	0.00796590	[8832] Tmp dir: 'C:\Users\abhishek\AppData\Local\Temp\' 	
00000008	0.00809910	[8832] AppStore version detected 	
00000009	0.01781500	[8832] Binary executable is signed by 'Akeo Consulting' 	
00000010	0.01784610	[8832] Will use settings from registry 	
00000011	0.01801130	[8832] found locale 'en-US' 	
00000012	0.05339040	[8832] Notice: Disabling VDS (Could not load VDS Service: [0x80070005] Access is denied.) 	
00000013	0.05448790	[8832] loc file not found in current directory - embedded one will be used 	
00000014	0.05568620	[8832] localization: extracted data to 'C:\Users\abhishek\AppData\Local\Temp\Ruf9BEB.tmp' 	
00000015	0.06468490	[8832] localization: found locale 'en-US' 	
00000016	0.06515880	[8832] localization: found locale 'ar-SA' 	
00000017	0.06554640	[8832] localization: found locale 'bg-BG' 	
00000018	0.06597800	[8832] localization: found locale 'zh-CN' 	
00000019	0.06631320	[8832] localization: found locale 'zh-TW' 	
00000020	0.06663220	[8832] localization: found locale 'hr-HR' 	
00000021	0.06696370	[8832] localization: found locale 'cs-CZ' 	
00000022	0.06731080	[8832] localization: found locale 'da-DK' 	
00000023	0.06764310	[8832] localization: found locale 'nl-NL' 	
00000024	0.06799280	[8832] localization: found locale 'fi-FI' 	
00000025	0.06833700	[8832] localization: found locale 'fr-FR' 	
00000026	0.06870730	[8832] localization: found locale 'de-DE' 	
00000027	0.06905230	[8832] localization: found locale 'el-GR' 	
00000028	0.06949490	[8832] localization: found locale 'he-IL' 	
00000029	0.06988000	[8832] localization: found locale 'hu-HU' 	
00000030	0.07023830	[8832] localization: found locale 'id-ID' 	
00000031	0.07057070	[8832] localization: found locale 'it-IT' 	
00000032	0.07095260	[8832] localization: found locale 'ja-JP' 	
00000033	0.07133490	[8832] localization: found locale 'ko-KR' 	
00000034	0.07169970	[8832] localization: found locale 'lv-LV' 	
00000035	0.07203830	[8832] localization: found locale 'lt-LT' 	
00000036	0.07238590	[8832] localization: found locale 'ms-MY' 	
00000037	0.07272370	[8832] localization: found locale 'nb-NO' 	
00000038	0.07306310	[8832] localization: found locale 'fa-IR' 	
00000039	0.07349870	[8832] localization: found locale 'pl-PL' 	
00000040	0.07385950	[8832] localization: found locale 'pt-BR' 	
00000041	0.07420760	[8832] localization: found locale 'pt-PT' 	
00000042	0.07456220	[8832] localization: found locale 'ro-RO' 	
00000043	0.07490600	[8832] localization: found locale 'ru-RU' 	
00000044	0.07533190	[8832] localization: found locale 'sr-RS' 	
00000045	0.07567040	[8832] localization: found locale 'sk-SK' 	
00000046	0.07602880	[8832] localization: found locale 'sl-SI' 	
00000047	0.07637000	[8832] localization: found locale 'es-ES' 	
00000048	0.07672210	[8832] localization: found locale 'sv-SE' 	
00000049	0.07705440	[8832] localization: found locale 'th-TH' 	
00000050	0.07751980	[8832] localization: found locale 'tr-TR' 	
00000051	0.07787400	[8832] localization: found locale 'uk-UA' 	
00000052	0.07828240	[8832] localization: found locale 'vi-VN' 	
00000053	0.07981800	[8832] Note: UAC is active 	
00000054	0.16243260	[8832] localization: initializing default message table 	
00000055	0.26995641	[8832] Rufus x64 v4.4.2103 (AppStore version) 	
00000056	0.27432621	[8832] Windows version: Windows 11 Pro x64 (Build 22631.3085) 	
00000057	0.27601650	[8832] Syslinux versions: 4.07/2013-07-25, 6.04/pre1 	
00000058	0.27936289	[8832] Grub versions: 0.4.6a, 2.06 	
00000059	0.28009000	[8832] System locale ID: 0x0809 (en-GB) 	
00000060	0.28059879	[8832] Will use default UI locale 0x0809 	
00000061	0.28059879	[8832] SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E 	
00000062	0.28059879	[8832] Localization set to 'en-US' 	
00000063	0.28117210	[8832] Found 517 officially revoked UEFI bootloaders from embedded list 	
00000064	0.28876111	[8832] Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b 	
00000065	0.28971341	[8832] Notice: Windows VDS is unavailable 	
00000066	0.46474639	[8832] Found USB 2.0 device 'Sony Storage Media USB Device' (054C:05BA) 	
00000067	0.92781061	[8832] 1 device found 	
00000068	1.01648831	[8832] Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes 	
00000069	1.01787055	[8832] Cylinders: 942, Tracks per cylinder: 255, Sectors per track: 63 	
00000070	1.01863670	[8832] Partition type: GPT, NB Partitions: 1 	
00000071	1.01933765	[8832] Disk GUID: {8A1943B3-F8E6-4D77-8A40-B3AFE9BB506B} 	
00000072	1.02015805	[8832] Max parts: 128, Start Offset: 17408, Usable = 7751039488 bytes 	
00000073	1.02082455	[8832] Partition 1: 	
00000074	1.02082455	[8832]   Type: Microsoft Basic Data Partition 	
00000075	1.02148676	[8832]   Name: 'Main Data Partition' 	
00000076	1.02401841	[8832]   Detected File System: NTFS 	
00000077	1.02456546	[8832]   ID: {05E8E477-94FF-4DA5-A308-18B22A664EF0} 	
00000078	1.02456546	[8832]   Size: 7.2 GB (7750004736 bytes) 	
00000079	1.02456546	[8832]   Start Sector: 2048, Attributes: 0x0000000000000000 	
00000080	7.63320065	[8832] Scanning image... 	
00000081	7.65086222	[8832] ISO analysis: 	
00000082	7.65281534	[8832]   Image is a UDF image 	
00000083	7.69254017	[8832] Disk image analysis: 	
00000084	7.69435263	[8832]   Image does not have a Boot Marker 	
00000085	7.73979521	[8832]   Could not access image: [0x0000000D] The data is invalid. 	
00000086	7.74172258	[8832] Could not acquire WIM index 	
00000087	7.74927187	[8832] ISO label: 'CCCOMA_X64FRE_EN-GB_DV9' 	
00000088	7.74973583	[8832]   Size: 6.3 GB (Projected) 	
00000089	7.75014400	[8832]   Has a >4GB file 	
00000090	7.75049925	[8832]   Uses: EFI 	
00000091	7.75085831	[8832]   Uses: Bootmgr (BIOS and UEFI) 	
00000092	7.75122023	[8832]   Uses: Install.wim (version 0.13.1) 	
00000093	7.75213671	[8832] Using image: Win11_23H2_EnglishInternational_x64v2.iso (6.3 GB) 		

```

pbatard, 2024-02-05:
Thanks for the report and especially the log from DebugView.

What I am seeing from your log seems to indicate that either your image is corrupted or your Windows installation is preventing Rufus to access some data that it needs to function properly:
```
00000085	7.73979521	[8832]   Could not access image: [0x0000000D] The data is invalid. 	
00000086	7.74172258	[8832] Could not acquire WIM index 	
```

The first thing I would do then is post the SHA-256 of the image you are using here so that it can be validated, because, while you say that _"The file has been downloaded from Microsoft's official website (...). The image does not seem tampered"_ is is possible for downloads to become corrupted, so I would really like you to perform the operation indicated in the checklist at:

> - [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Also, you are stating that the application simply closes rather than crashes? Do you get any prompt to send data to Microsoft when that happens (as you would with an application crash)? Because if you do, and since it's the AppStore version, I can ultimately access that data so I might be able to have a closer look at what really happens.

I would also suggest, since you are using the AppStore version, to try with the standalone Rufus application, which you can download from https://rufus.ie to see if you experience the same issue. For the record, I tested Rufus by force triggering the `Could not access image` / `Could not acquire WIM index` errors, but I was still able to proceed successfully without a crash or the application closing, so this looks like an environmental/image issue rather than a Rufus bug.

abhi-deshpande, 2024-02-05:
Thanks for your reply..!

1. I calculated the checksums for the ISO file I am using and got this in the log:
```
Using image: Win11_23H2_EnglishInternational_x64v2.iso (6.3 GB)

Computing hash for 'D:\Win11_23H2_EnglishInternational_x64v2.iso'...
  MD5:    8095893e343b797ce83ef87af1f076c2
  SHA1:   10793cd70013dc4d5097cf30f2e9f2ba2a4865f5
  SHA256: d989bd2572fb7aeec6d10d5abec668dbcf714b6b9f4a89de7bf2409973e4a3e7
```

2. Rufus does not crash. It just closes without any hints or log entries. Windows does not prompt me to report any errors, or crashes. I tried the same procedure with the standalone version of Rufus and did not find any different behaviour other than this same issue.

3. As I stated in the OP, Rufus worked when I changed partition scheme to MBR. Unfortunately, my computer went to sleep during the writing process that's why it was interrupted and later I cancelled it. But what's more important is it worked..!  I found this in the log:
```
00000001	0.00000000	[3300] *** Rufus init *** 	
00000002	0.00516260	[3300] Cur dir: 'D:\Software Executables\' 	
00000003	0.00529040	[3300] App dir: 'D:\Software Executables\' 	
00000004	0.00539830	[3300] Sys dir: 'C:\Windows\system32' 	
00000005	0.00543300	[3300] Usr dir: 'C:\Users\abhishek' 	
00000006	0.00546000	[3300] Dat dir: 'C:\Users\abhishek\AppData\Local' 	
00000007	0.00547300	[3300] Tmp dir: 'C:\Users\abhishek\AppData\Local\Temp\' 	
00000008	0.01208450	[3300] Binary executable is signed by 'Akeo Consulting' 	
00000009	0.01219490	[3300] Will use settings from registry 	
00000010	0.01228750	[3300] found locale 'en-US' 	
00000011	0.06299260	[3300] loc file not found in current directory - embedded one will be used 	
00000012	0.06382110	[3300] localization: extracted data to 'C:\Users\abhishek\AppData\Local\Temp\Ruf1115.tmp' 	
00000013	0.06867930	[3300] localization: found locale 'en-US' 	
00000014	0.06900100	[3300] localization: found locale 'ar-SA' 	
00000015	0.06931390	[3300] localization: found locale 'bg-BG' 	
00000016	0.06964410	[3300] localization: found locale 'zh-CN' 	
00000017	0.06995720	[3300] localization: found locale 'zh-TW' 	
00000018	0.07025740	[3300] localization: found locale 'hr-HR' 	
00000019	0.07053610	[3300] localization: found locale 'cs-CZ' 	
00000020	0.07081610	[3300] localization: found locale 'da-DK' 	
00000021	0.07110170	[3300] localization: found locale 'nl-NL' 	
00000022	0.07138480	[3300] localization: found locale 'fi-FI' 	
00000023	0.07167180	[3300] localization: found locale 'fr-FR' 	
00000024	0.07197900	[3300] localization: found locale 'de-DE' 	
00000025	0.07226790	[3300] localization: found locale 'el-GR' 	
00000026	0.07259810	[3300] localization: found locale 'he-IL' 	
00000027	0.07291210	[3300] localization: found locale 'hu-HU' 	
00000028	0.07320710	[3300] localization: found locale 'id-ID' 	
00000029	0.07348600	[3300] localization: found locale 'it-IT' 	
00000030	0.07381820	[3300] localization: found locale 'ja-JP' 	
00000031	0.07412710	[3300] localization: found locale 'ko-KR' 	
00000032	0.07453330	[3300] localization: found locale 'lv-LV' 	
00000033	0.07487210	[3300] localization: found locale 'lt-LT' 	
00000034	0.07518090	[3300] localization: found locale 'ms-MY' 	
00000035	0.07546850	[3300] localization: found locale 'nb-NO' 	
00000036	0.07575140	[3300] localization: found locale 'fa-IR' 	
00000037	0.07608540	[3300] localization: found locale 'pl-PL' 	
00000038	0.07636920	[3300] localization: found locale 'pt-BR' 	
00000039	0.07665160	[3300] localization: found locale 'pt-PT' 	
00000040	0.07694430	[3300] localization: found locale 'ro-RO' 	
00000041	0.07723430	[3300] localization: found locale 'ru-RU' 	
00000042	0.07755870	[3300] localization: found locale 'sr-RS' 	
00000043	0.07784680	[3300] localization: found locale 'sk-SK' 	
00000044	0.07813350	[3300] localization: found locale 'sl-SI' 	
00000045	0.07841680	[3300] localization: found locale 'es-ES' 	
00000046	0.07870880	[3300] localization: found locale 'sv-SE' 	
00000047	0.07898500	[3300] localization: found locale 'th-TH' 	
00000048	0.07933250	[3300] localization: found locale 'tr-TR' 	
00000049	0.07962690	[3300] localization: found locale 'uk-UA' 	
00000050	0.07995290	[3300] localization: found locale 'vi-VN' 	
00000051	0.08130000	[3300] Note: UAC is active 	
00000052	0.09964000	[3300] localization: initializing default message table 	
00000053	0.26366770	[3300] Rufus x64 v4.4.2103 	
00000054	0.26370040	[3300] Windows version: Windows 11 Pro x64 (Build 22631.3085) 	
[... 445 lines cut ...]
00000500	1846.90771484	[3300] Cancelling 	
00000501	1852.92163086	[3300] Blocking I/O operation detected 	
00000502	1883.08996582	[3300]  	
00000503	1883.17224121	[3300] Found USB 2.0 device 'Sony Storage Media USB Device' (054C:05BA) 	
00000504	1883.70654297	[3300] 1 device found 	
00000505	1883.81555176	[3300] Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes 	
00000506	1883.81567383	[3300] Cylinders: 942, Tracks per cylinder: 255, Sectors per track: 63 	
00000507	1883.81616211	[3300] Partition type: MBR, NB Partitions: 2 	
00000508	1883.81640625	[3300] Disk ID: 0x01AC5D7F 	
00000509	1883.82141113	[3300] Drive has a Rufus Master Boot Record 	
00000510	1883.82189941	[3300] Partition 1: 	
00000511	1883.82275391	[3300]   Type: NTFS (0x07) 	
00000512	1883.82275391	[3300]   Detected File System: NTFS 	
00000513	1883.82275391	[3300]   Size: 7.2 GB (7748952064 bytes) 	
00000514	1883.82275391	[3300]   Start Sector: 2048, Boot: Yes 	
[... cut ...]
00000516	1883.83251953	[3300]   Type: EFI System Partition (0xef) 	
00000517	1883.83251953	[3300]   Detected File System: FAT12 	
00000518	1883.83251953	[3300]   Size: 1 MB (1048576 bytes) 	
00000519	1883.83251953	[3300]   Start Sector: 15136720, Boot: No 	
00000520	1885.92089844	[3300] Killed blocking I/O timer 	
00000521	1918.22460938	[3300] Saved rufus.log 	
00000522	1918.61621094	[3300] SetLGP: Successfully restored NoDriveTypeAutorun policy to 0x0000009E 	
00000523	1918.61743164	[3300] *** Rufus exit *** 	
```

pbatard, 2024-02-05:
Well, that SHA-1 does __NOT__ match the SHA-1 of any official Windows ISOs. See https://github.com/pbatard/rufus/wiki/FAQ#user-content-How_can_I_validate_that_a_Windows_ISO_is_a_genuine_retail_version and especially https://sha1.rg-adguard.net/search.php?sha1=10793cd70013dc4d5097cf30f2e9f2ba2a4865f5 returns no hit.

The SHA-1 for `Win11_23H2_EnglishInternational_x64v2.iso` downloaded from Microsoft should be `4138eca827754ba308e4a3eb8d11e86ae5331369`.

So there is definitely a problem with your ISO to start with. Either it is not an official ISO, or your download got corrupted.

The first thing you need to do then is download `Win11_23H2_EnglishInternational_x64v2.iso` again (which, by the way, you can also do [using Rufus](https://raw.githubusercontent.com/wiki/pbatard/rufus/images/download.gif)) and make sure that the SHA-1 reports the value above.

And yes, I did read that it worked when switching to MBR, but I am interested in finding out the cause of the issue for Rufus exiting, which is definitely linked to either your ISO or your environment. The fact that you don't even get any of the exit messages in DebugView indicates that the application actually crashes rather than exits (there's no code in Rufus that makes the application just bail out). And I'm going to assume that you did try other Windows ISOs in GPT mode, and did not experience the crash? Or does this happen with all the Windows ISOs you tried?

At any rate, knowing what happens with a valid ISO will tell us more (but please keep a copy of that problematic ISO just in case).

abhi-deshpande, 2024-02-06:
I downloaded that ISO once again. This time I did verify the checksums properly. Rufus did not crash this time. It worked as expected, and showed the installation customization options. As you pointed out, the issue is indeed related to the corrupt ISO file. I think this issue can be closed. Thank you for your support..!

[image]

pbatard, 2024-02-06:
Thanks for reporting back. I'll close this issue then.