pbatard/rufus #2426: Bug: Rufus isn't supporting WIM files

Videogamer555, 2024-02-06:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When selecting a WIM type image, it ALWAYS pops up the error box that says "This image is either non-bootable, or it uses a boot or compression method that is not supported by Rufus...". This happens for booth boot.wim and install.wim when I attempt to select it as the image file to be used in Rufus. It happens regardless of whether I select "Disk or ISO image" or "Non bootable" from the "Boot selection" dropdown list. The log from Rufus provides a bit more info on the issue where it says "Image is a NON-UEFI bootable Windows 7 (Build 7601.17514) installation image" (note that the complete log can be seen in the correct section below this paragraph). So it actually does recognize that it's a correct installation image, but seems to have a problem with it being a NON-UEFI bootable image. I thought that Rufus supported BOTH types of booting (UEFI booting, and also legacy BIOS booting), but the log seems to indicate that Rufus will refuse to work with any image that's not for UEFI booting. Has Rufus changed recently to not accept WIM files intended for legacy BIOS booting? Or is this some kind of bug?

Log
---
```
Rufus x64 v4.4.2103
Windows version: Windows 11 Pro x64 (Build 22621.3007)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
0 devices found
Scanning image...
ISO analysis:
  'C:\Users\benhu\AppData\Local\Temp\boot.wim' doesn't look like an ISO image
Disk image analysis:
  Image does not have a Boot Marker
  Image is a NON-UEFI bootable Windows 7 (Build 7601.17514) installation image
Scanning image...
ISO analysis:
  'C:\Users\benhu\AppData\Local\Temp\install.wim' doesn't look like an ISO image
Disk image analysis:
  Image does not have a Boot Marker
  Image is a NON-UEFI bootable Windows 7 (Build 7601.17514) installation image

```

JonnyTech, 2024-02-06:
From the website:

`Rufus is a utility that helps format and create bootable USB flash drives ... from bootable ISOs (Windows, Linux, UEFI, etc.)`

And the project page:

`Rufus is a utility that helps format and create bootable USB flash drives ... Create bootable drives from bootable ISOs (Windows, Linux, etc.)`

Where does it state that you can also do so from WIM files?

closed 2024-02-06 by pbatard

Videogamer555, 2024-02-07:
> @Videogamer555 <https://github.com/Videogamer555>,
>
> The log line you quoted pretty much gives you the answer:
>
> Image is a NON-UEFI bootable Windows 7 (Build 7601.17514) installation image
>
> Rufus can only open WIM files that contain UEFI bootloaders at the
> expected UEFI standard location (because, otherwise, it will obvioulsy not
> be able to create bootable media from that). The image you are trying to
> use does not.
>
> Furthermore, since your WIM is a Windows 7 WIM:
> https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_XP_Windows_Vista_Windows_7_NO_LONGER_SUPPORTED

I'm not trying to run it on Windows 7. I'm trying to run it on Windows 10,
but use a Windows 7 WIM file to make a USB drive boot to Windows 7, so I
can use it as an alternate OS. Boot from the harddrive you get Win 10. Boot
from the USB and you get Win 7. That's my intent. By the way, not all WIM
files (or other disk images for that matter), are designed for UEFI boot.
Many are designed for BIOS boot instead. Is Rufus now going to drop all
support for BIOS booted OSes?

Videogamer555, 2024-02-07:
I also didn't understand what you meant in the page you linked, that you
don't have time to continue to provide support for Windows 7 or XP. It
seems to me that once the feature was already in Rufus, that it then would
just be in Rufus. It actually seems like it would take more time for you to
actively REMOVE support for those older OSes. Did Microsoft contact you and
demand you remove support for them, under threat of lawsuit if you didn't
remove support for those older OSes?

pineapple63, 2024-02-07:
From my understanding, the not supporting Windows 7 or older is that there are no plans to remove that functionality from the software, but if you run into problems, do not expect to receive help
I think the issue is that for Windows 7 you need to use a complete ISO file (regardless of the OS that is running on the computer you are using)

pbatard, 2024-02-07:
> but use a Windows 7 WIM file to make a USB drive boot to Windows 7

What part of _"Please don't ask for support regarding Windows XP, Windows Vista or Windows 7 installations"_ is difficult to understand? If you are trying to boot to Windows 7 in any way shape or form, you are 100% on your own.

> It seems to me that once the feature was already in Rufus, that it then would just be in Rufus.

Once a feature is in Rufus, it is expected to be __maintained__ so that it continues to work as expected. Maintenance is not a zero cost operation. In fact, the vast majority of time I spend on Rufus is spent on maintenance. The code of Rufus is constantly being modified, and some of these modifications apply to the parts of the code that enabled it to work on Windows 7. I'm not going to provide a half-assed user experience on any platform period. Therefore I will deliberately remove the platforms that I no longer have the resources to test. 

See also https://old.reddit.com/r/windows7/comments/18g65j9/rufus_43_running_on_windows_7/kd1q2or/ which is relevant to your question.