pbatard/rufus #2429: Rufus stuck when downloading Win11

jsb-avf, 2024-02-07:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When I want to download a Win11 image with Rufus, Rufus gets stuck when executing the download script and does not continue. In the log Rufus gets to the point `Querying https://www.microsoft.com/de-DE/software-download/', nothing more happens even after more than 20 minutes. I have tried to change the language to English, but Rufus also gets stuck at https://www.microsoft.com/us-us/software-download/. 

Log
---
```Rufus x64 v4.4.2103 (Portable)
Windows version: Windows 10 Pro x64 (Build 19045.3930)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
0 devices found
Downloading https://github.com/pbatard/Fido/releases/download/v1.54/Fido.ps1.lzma
File length: 17.4 KB
Successfully downloaded 'Fido.ps1.lzma'
Download signature is valid ✓
Script signature is valid ✓
Please Wait...
Querying https://www.microsoft.com/de-DE/software-download/
```

pbatard, 2024-02-07:
This should be fixed now.

Please try again and makes sure it says `Downloading https://github.com/pbatard/Fido/releases/download/v1.55/Fido.ps1.lzma` in the log (if it doesn't, it means you're still using the v1.54 cached version and you will need to try until you see `v1.55`).

jsb-avf, 2024-02-07:
I can confirm, the bug is fixed. Thanks!

closed 2024-02-07 by jsb-avf