pbatard/rufus #2430: Feature request: Add menu which disable Windows Defender when creatiing a ISO with Windows 10/11.

ionuttbara, 2024-02-12:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
Issue description
-----------------
As <a href="https://github.com/ionuttbara/windows-defender-remover">Defender Remover</a> dev, i have a suggestion for you. Add posibility to disable Defender when create a USB Drive with Windows ISO (Windows 10 and 11) with Defender Antivirus disabled by default. 
You have an menu which creates a file named **unattended.xml** which is have configurations such custom regional settings and disablation of bitlocker.
To disable antivirus entirely you must disable following services: *Sense,WdBoot,WdFilter,WdNisDrv,WdNisSvc and WinDefend*

Here is the XML code part which is disables Defender in **unattend.xml**, and starts from **<component name="Microsoft-Windows-Deployment" processorArchitecture="amd64" publicKeyToken="31bf3856ad364e35" language="neutral" versionScope="nonSxS">**.
```
<?xml version="1.0" encoding="utf-8"?>
<unattend xmlns="urn:schemas-microsoft-com:unattend" xmlns:wcm="http://schemas.microsoft.com/WMIConfig/2002/State">
	<!--https://schneegans.de/windows/unattend-generator/?LanguageMode=Interactive&ProcessorArchitecture=amd64&ComputerNameMode=Random&TimeZoneMode=Implicit&PartitionMode=Interactive&WindowsEditionMode=Interactive&UserAccountMode=Interactive&LockoutMode=Default&DisableDefender=true&WifiMode=Interactive&ExpressSettings=Interactive&WdacMode=Skip-->
	<settings pass="offlineServicing"></settings>
	<settings pass="windowsPE">
		<component name="Microsoft-Windows-Setup" processorArchitecture="amd64" publicKeyToken="31bf3856ad364e35" language="neutral" versionScope="nonSxS">
			<UserData>
				<ProductKey>
					<Key>00000-00000-00000-00000-00000</Key>
				</ProductKey>
				<AcceptEula>true</AcceptEula>
			</UserData>
		</component>
	</settings>
	<settings pass="generalize"></settings>
	<settings pass="specialize">
		<component name="Microsoft-Windows-Deployment" processorArchitecture="amd64" publicKeyToken="31bf3856ad364e35" language="neutral" versionScope="nonSxS">
			<RunSynchronous>
				<RunSynchronousCommand wcm:action="add">
					<Order>1</Order>
					<Path>cmd.exe /c "&gt;&gt;"%TEMP%\regini.txt" echo HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\Sense"</Path>
				</RunSynchronousCommand>
				<RunSynchronousCommand wcm:action="add">
					<Order>2</Order>
					<Path>cmd.exe /c "&gt;&gt;"%TEMP%\regini.txt" echo     "Start" = REG_DWORD 4"</Path>
				</RunSynchronousCommand>
				<RunSynchronousCommand wcm:action="add">
					<Order>3</Order>
					<Path>cmd.exe /c "&gt;&gt;"%TEMP%\regini.txt" echo HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\WdBoot"</Path>
				</RunSynchronousCommand>
				<RunSynchronousCommand wcm:action="add">
					<Order>4</Order>
					<Path>cmd.exe /c "&gt;&gt;"%TEMP%\regini.txt" echo     "Start" = REG_DWORD 4"</Path>
				</RunSynchronousCommand>
				<RunSynchronousCommand wcm:action="add">
					<Order>5</Order>
					<Path>cmd.exe /c "&gt;&gt;"%TEMP%\regini.txt" echo HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\WdFilter"</Path>
				</RunSynchronousCommand>
				<RunSynchronousCommand wcm:action="add">
					<Order>6</Order>
					<Path>cmd.exe /c "&gt;&gt;"%TEMP%\regini.txt" echo     "Start" = REG_DWORD 4"</Path>
				</RunSynchronousCommand>
				<RunSynchronousCommand wcm:action="add">
					<Order>7</Order>
					<Path>cmd.exe /c "&gt;&gt;"%TEMP%\regini.txt" echo HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\WdNisDrv"</Path>
				</RunSynchronousCommand>
				<RunSynchronousCommand wcm:action="add">
					<Order>8</Order>
					<Path>cmd.exe /c "&gt;&gt;"%TEMP%\regini.txt" echo     "Start" = REG_DWORD 4"</Path>
				</RunSynchronousCommand>
				<RunSynchronousCommand wcm:action="add">
					<Order>9</Order>
					<Path>cmd.exe /c "&gt;&gt;"%TEMP%\regini.txt" echo HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\WdNisSvc"</Path>
				</RunSynchronousCommand>
				<RunSynchronousCommand wcm:action="add">
					<Order>10</Order>
					<Path>cmd.exe /c "&gt;&gt;"%TEMP%\regini.txt" echo     "Start" = REG_DWORD 4"</Path>
				</RunSynchronousCommand>
				<RunSynchronousCommand wcm:action="add">
					<Order>11</Order>
					<Path>cmd.exe /c "&gt;&gt;"%TEMP%\regini.txt" echo HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\WinDefend"</Path>
				</RunSynchronousCommand>
				<RunSynchronousCommand wcm:action="add">
					<Order>12</Order>
					<Path>cmd.exe /c "&gt;&gt;"%TEMP%\regini.txt" echo     "Start" = REG_DWORD 4"</Path>
				</RunSynchronousCommand>
				<RunSynchronousCommand wcm:action="add">
					<Order>13</Order>
					<Path>regini.exe "%TEMP%\regini.txt"</Path>
				</RunSynchronousCommand>
			</RunSynchronous>
		</component>
	</settings>
	<settings pass="auditSystem"></settings>
	<settings pass="auditUser"></settings>
	<settings pass="oobeSystem">
		<component name="Microsoft-Windows-Shell-Setup" processorArchitecture="amd64" publicKeyToken="31bf3856ad364e35" language="neutral" versionScope="nonSxS">
			<OOBE>
				<HideEULAPage>true</HideEULAPage>
				<HideWirelessSetupInOOBE>false</HideWirelessSetupInOOBE>
			</OOBE>
		</component>
	</settings>
</unattend>
        ```
        Note: You'll see in code order of command from 1 to 13, these command must be in that order.
        Optional mentions: This

JonnyTech, 2024-02-12:
@ionuttbara are you suggesting to create installable media that installs Windows onto a user's computer with a **core safety feature** disabled or removed, thus putting said user at risk and vulnerable to malware? Or am I misunderstanding your reasoning?

ionuttbara, 2024-02-12:
depend in what context if you talk about.
for example, if you want to install a 3rd antivirus provider or test machine. And this is optional, a check box in iso creation.

JonnyTech, 2024-02-12:
That is rather an edge case.

See item 3 here: [Why can't you just add the feature I want?](https://github.com/pbatard/rufus/wiki/FAQ#why-cant-you-just-add-the-feature-i-want)

ionuttbara, 2024-02-12:
now, waiting for dev. rufus provides disablation of automatic encyption (biitlocker) and this can be another example. if the feature will not added, i will add in DefenderRemover that.

pbatard, 2024-02-12:
This has been suggested before, and I have no plans to add this. Please see https://github.com/pbatard/rufus/wiki/FAQ#user-content-Could_you_add_support_for_removing_ltWindows_option_Xgt_in_the_Windows_User_Experience_dialog.

The goal of the Windows User Experience in Rufus is not to add a checkbox for every Windows configurable feature under the sun (because if I were to do so, the user would be presented with a dialog listing hundreds of options) but __only__ for the ones that prevent the installation process from being completed/streamlined.

In short, the goal I have with the Windows User Experience options I add to Rufus is to enable someone to plug a Windows installation drive in a computer, launch the installation, go away, and come back to a completed installation of Windows that they can start to use (or configure further with regards to features that do not halt the installation process). That's it.

Considering that the presence of Defender does not hinder installation, and that its removal can be enacted by the user post install, the setting you suggest will not be added to WUE.

closed 2024-02-12 by pbatard