pbatard/rufus #2434: Error installing windows 11 when using custom options in rufus

Sxvn1337, 2024-02-19:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I am using rufus to create the usb, and in rufus I use the options
- bypass tpm check
- bypass ms account check
- disable data collection
- disable bitlocker

The laptop I'm downloading to already (had) windows 11 home, so when I booted it, I got to the boot menu and booted from the usb and formatted the drive so that the old windows 11 was deleted. when I try to perform the new installation, after selecting edition and installing windows, during restart process I get the error "Windows could not parse or process the unattend answer file for pass
[specialize]. The settings specified in the answer file cannot be applied. The error was detected while processing settings for component [Microsoft-Windows-Deployment]."

However, when I disabled those options and then flashed the image, windows install went perfectly fine.

Expected behaviour
----------------------
Windows installs okay, everything works fine.

Actual behaviour
-------------------
Error described above.

Log
---
```
https://go.0xfc.de/39uqhw
```

pbatard, 2024-02-19:
Thanks for the report.

I have just tried to replicate the issue using the same options as the ones you used, and went through a full installation of Windows 11 23H2v2, but I never saw the error you mention.

If you still have the drive you created that produced the error, could you please copy/paste the `sources\$OEM$\$$\Panther\unattend.xml` file you will find there? Or if not, could you recreate the media in the same way as you created it and then copy that file?

Sxvn1337, 2024-02-19:
https://go.0xfc.de/dv5tfp

pbatard, 2024-02-19:
Yes, this is the exact same content as the one I have, so I am very surprised this answer file would generate an error on your side (because **this** is the unattend answer file that your setup says it cannot parse, and specifically, the `BypassNRO` section) but not on mine. This means that there has to be some sort of environmental component to the issue.

I'm going to try a few more installs, but it looks to me that, under some circumstances, and some machines, `BypassNRO` will be rejected by Windows setup and generate an error. My guess is that it may have to do with remnants of a previous installation, or some networked licensing exchange between the Microsoft servers and your machine during the installation process, in which case you may want to make sure that your network is disconnected when you proceed with the install (which it **should** be anyway, as the only way `BypassNRO` will work in the first place is if Windows can **not** connect to a network). But outside of this, I don't really have much of any other ideas for now.

pbatard, 2024-02-19:
Still can't replicate the issue. I also suspect that this has to do with whether you entered a product key and what edition of Windows you selected for installation.

So, in order for me to investigate further I would like to know:
1. Did you enter of product key? Or did you select the _I don't have a product key_ option?
2. What edition of Windows did you select for installation?

Sxvn1337, 2024-02-20:
1. Never had the chance to input the product key. The laptop came with W11 Home edition in S mode. I flashed my thumb drive and then installed another edition of windows over it, (and then the whole issue happened) so I disabled all options and successfully got a new install of W11 on my laptop. Even in the installation screen, there was nothing about a product key. (Which is quite weird).
2. Windows 11 Pro Education edition

pbatard, 2024-02-20:
> Never had the chance to input the product key. The laptop came with W11 Home edition in S mode.

There we go. That's the issue.

Windows S Mode is a global setting, that is applied to the hardware in a permanent manner by Windows (it is stored in the UEFI variables, so even if you remove/clear Windows completely, the setting is persisted) and that __forces__ the use of a global account.

Therefore, if your run the installer on a machine that had Windows S installed, and you use the bypass global account option, then of course Windows setup will produce an error about trying to bypass that option.

So there's the issue.

Unfortunately, sin Rufus does **not** alter any executable files, it is not possible for Rufus to detect Windows S prior to installation, so, short of adding a specific entry in the FAQ about this (which I am planning to do) there's not much that can be done. You're just trying to make two incompatible options (Windows enforcement of a global user account on hardware where Windows S was installed vs. trying to disable global user account) work together, whereas they obviously can't.

oxynet12, 2024-02-20:
How fix

Sxvn1337, 2024-02-20:
@pbatard I was successfully able to remove S mode once. However on a clean install after that S mode still persists. Is the laptop forever locked to S mode?

pbatard, 2024-02-20:
> I was successfully able to remove S mode once. However on a clean install after that S mode still persists. 

Then you didn't actually remove S mode. If you do it properly, S Mode removal is permanent. If S Mode still persists, then you didn't remove it properly.

Sxvn1337, 2024-02-20:
> Then you didn't actually remove S mode. If you do it properly, S Mode removal is permanent. If S Mode still persists, then you didn't remove it properly.

Very weird. I logged in with my MS account, removed S mode... restarted and it wasn't there. Came back after the fresh install.

pbatard, 2024-02-21:
I have now created a new FAQ entry:
https://github.com/pbatard/rufus/wiki/FAQ#user-content-The_settings_specified_in_the_answer_file_cannot_be_applied

Since there is nothing more I can do about this, as, again, this is a pure incompatibility enacted by Windows Setup between a PC having been set to using S Mode (which is a permanent setting that is persisted in the hardware) and the enforcement of a Microsoft online account when S Mode is detected, I will close this issue.

closed 2024-02-21 by pbatard