pbatard/rufus #2436: USB drives become write protected

Hubalala, 2024-02-22:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

Yes yes all done and read from FAQ that Rufus is not damaging USB drives but excuse me - am I stupid or how else to understand thing happening infront of my eyes? Twice installation of windows 11 iso failed on 2 different drives which are working fine for years and last time yesterday.. sigh

Are these USB sticks fixable?

Rufus x64 v4.4.2103
Windows version: Windows 10 Pro x64 (Build 19045.4046)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
Disk type: Removable, Disk size: 7.9GB, Sector size: 512 bytes
Cylinders: 954, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {971C7E38-B729-46CF-9F8D-2FDDC6AAABB7}
Max parts: 128, Start Offset: 17408, Usable = 7851702784 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {4E6577B0-0910-4D1C-9035-2D8472E97AC2}
  Size: 7.3 GB (7849590784 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {941DD763-0C54-436A-90B2-92FF80E110E3}
  Size: 1 MB (1048576 bytes)
  Start Sector: 15333280, Attributes: 0x9000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-GB_DV9'
  Detected: Windows 11 ISO (Build 22621.2861)
  Size: 6.3 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win11_23H2_EnglishInternational_x64v2.iso (6.3 GB)
Selected Windows User Experience options:
• Bypass SB/TPM/RAM
• Bypass online account requirement
• Disable data collection
• Use 'Admin' for local account name
• Use the same regional options as this user's

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Warning: Could not delete partition(s): [0x8007045D] The request could not be performed because of an I/O device error.
Opened \\.\PhysicalDrive1 for shared write access
Requesting logical volume handle...
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Write error [0x0000045D] The request could not be performed because of an I/O device error.
Retrying in 5 seconds...
Write error [0x0000045D] The request could not be performed because of an I/O device error.
Retrying in 5 seconds...
Write error [0x0000045D] The request could not be performed because of an I/O device error.
Retrying in 5 seconds...
Write error [0x0000045D] The request could not be performed because of an I/O device error.
Could not reset partitions
Re-mounted volume as E: after error

Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
Disk type: Removable, Disk size: 7.9GB, Sector size: 512 bytes
Cylinders: 954, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {971C7E38-B729-46CF-9F8D-2FDDC6AAABB7}
Max parts: 128, Start Offset: 17408, Usable = 7851702784 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {4E6577B0-0910-4D1C-9035-2D8472E97AC2}
  Size: 7.3 GB (7849590784 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {941DD763-0C54-436A-90B2-92FF80E110E3}
  Size: 1 MB (1048576 bytes)
  Start Sector: 15333280, Attributes: 0x9000000000000000
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
Disk type: Removable, Disk size: 7.9GB, Sector size: 512 bytes
Cylinders: 954, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {971C7E38-B729-46CF-9F8D-2FDDC6AAABB7}
Max parts: 128, Start Offset: 17408, Usable = 7851702784 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {4E6577B0-0910-4D1C-9035-2D8472E97AC2}
  Size: 7.3 GB (7849590784 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {941DD763-0C54-436A-90B2-92FF80E110E3}
  Size: 1 MB (1048576 bytes)
  Start Sector: 15333280, Attributes: 0x9000000000000000

pbatard, 2024-02-22:
> how else to understand thing happening infront of my eyes?

I can postulate a few things: you were sold fake drive and only went over capacity when writing the large Windows ISO, so that's why they only failed when using Rufus. I have also had reports of people with flaky USB ports (if the power delivery is unstable for instance) that led to drive failure. Or these drives, which you don't provide the manufacturer and model name, were poorly manufacturers and do not support writing continuously at high speed.

There are many, **many**, ways your drive could have failed without involving Rufus and again, I invite everyone who *believes*, against evidence (especially, even if you are not technically versed in Windows internals, the very few reports of coincidental drive failures vs [lots of downloads](https://tooomm.github.io/github-release-stats/?username=pbatard&repository=rufus) should say something) that Rufus can damage drives to contact the manufacturers of the flash drives who were damaged and ask them to help corroborating these claims (because, if true, drives of the same model that experience a high rate of failure will be of __great interest__ for the manufacturer, to investigate the cause of failure, as it means they will ultimately save money on RMAs and/or bad reputation).

> Are these USB sticks fixable?

I/O device errors are indicative of a hardware failure so probably not.

pbatard, 2024-02-28:
No update &rarr; Closed.

closed 2024-02-28 by pbatard