pbatard/rufus #2440: U disk partition identifies the entire u disk

YuanJie2001, 2024-03-01:
hello,everybody.
My u disk has a pe boot partition and an ordinary file storage partition. I divide the free space of ordinary files into a third zone. I hope to use the third area to make a linux boot disk to install the linux system. But when burning, it recognizes the entire u-disk system and cannot select one of the partitions to make the linux startup program.Of course, I have also tried floppy disk access and ventoy.They're the same.
[image]

pineapple63, 2024-03-01:
The reason the software doesn't recognise individual partitions, is because the software has to format the drive in order to ensure it will boot (especially with older systems)

pbatard, 2024-03-01:
See also https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_does_Rufus_need_to_delete_all_partitionsvolumes_when_creating_a_drive

If you need software that preserves partition while creating bootable media, I will kindly ask that you use something else than Rufus, as it will never be designed for your usage scenario.

closed 2024-03-01 by pbatard