pbatard/rufus #2441: MAD BooTsticks through OS on creation?!

weTHEeffff, 2024-03-05:
Could it be possible, that my compromised OS is changing what is written to usb (instead of clean creation from iso)
-----------------
I ve got 2 laptops, and wanted to clean (re-)install the old dell e7250
I tried it with win10 ubuntu and archlinux installations with verified images, but it seems that anyhow all bootsticks are compromised.. (because even offline installs showed strange proccesses, auto user/permission creations (and then hiding them from userview), malicious trys on firmwareupdates (with crazy unreadable online destinations which are offline not reachable), scripts for changing/removing logs on linux and many couriouse things more 
(even if malewarebytes forum says logs  look clean on the Win11 Asus device where i created the sticks with rufus, I cant believe that..)

please have a look on the rufus creation log and tell me if this looks really clean to you before I try another install
so I can be sure that the problem is not related to the boot stick creation :')
thank you very much in advance for your time and expertise!

Log
---
```
Could it be possible, that my compromised OS is changing what is written to usb (instead of clean creation from iso)
-----------------
I ve got 2 laptops, and wanted to clean (re-)install the old dell e7250
I tried it with win10 ubuntu and archlinux installations with verified images, but it seems that anyhow all bootsticks are compromised.. (because even offline installs showed strange proccesses, auto user/permission creations (and then hiding them from userview), malicious trys on firmwareupdates (with crazy unreadable online destinations which are offline not reachable), scripts for changing/removing logs on linux and many couriouse things more 
(even if malewarebytes forum says logs  look clean on the Win11 Asus device where i created the sticks with rufus, I cant believe that..)

please have a look on the rufus creation log and tell me if this looks really clean to you before I try another install
so I can be sure that the problem is not related to the boot stick creation :')
thank you very much in advance for your time and expertise!

Log
---
```
Rufus x64 v4.4.2103
Windows version: Windows 11 Pro x64 (Build 22631.3155)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'USB Device' (058F:6387)
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 949, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x8CC6F18C
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI System Partition (0xef)
  Detected File System: FAT16
  Size: 15 MB (15728640 bytes)
  Start Sector: 1898496, Boot: No
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-GB_DV9'
  Detected: Windows 10 ISO (Build 19041.2965)
  Size: 5.7 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_22H2_EnglishInternational_x64v1.iso (5.7 GB)
Selected Windows User Experience options:
• Disable data collection
• Disable bitlocker

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'D:' as volume mountpoint
Opened \\.\PhysicalDrive2 for shared write access
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 7.3 GB)
● Creating UEFI:NTFS Partition (offset: 7810805760, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive2 for shared write access
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{15074bc1-757e-4d2d-a257-a8a52b3cd805}\
Disabling file indexing...
Successfully remounted \\?\Volume{15074bc1-757e-4d2d-a257-a8a52b3cd805}\ as D:
Extracting files...
Image is a UDF image
Extracting: D:\autorun.inf (128 bytes)
Extracting: D:\boot\bcd (16 KB)
Extracting: D:\boot\boot.sdi (3 MB)
Extracting: D:\boot\bootfix.bin (1 KB)
Extracting: D:\boot\bootsect.exe (107.8 KB)
Extracting: D:\boot\en-gb\bootsect.exe.mui (16 KB)
Extracting: D:\boot\etfsboot.com (4 KB)
Extracting: D:\boot\fonts\chs_boot.ttf (3.5 MB)
Extracting: D:\boot\fonts\cht_boot.ttf (3.7 MB)
Extracting: D:\boot\fonts\jpn_boot.ttf (1.9 MB)
Extracting: D:\boot\fonts\kor_boot.ttf (2.3 MB)
Extracting: D:\boot\fonts\malgun_boot.ttf (174.2 KB)
Extracting: D:\boot\fonts\malgun_console.ttf (79.5 KB)
Extracting: D:\boot\fonts\malgunn_boot.ttf (171.8 KB)
Extracting: D:\boot\fonts\meiryo_boot.ttf (141.3 KB)
Extracting: D:\boot\fonts\meiryo_console.ttf (88.5 KB)
Extracting: D:\boot\fonts\meiryon_boot.ttf (139.7 KB)
Extracting: D:\boot\fonts\msjh_boot.ttf (157.6 KB)
Extracting: D:\boot\fonts\msjh_console.ttf (198.1 KB)
Extracting: D:\boot\fonts\msjhn_boot.ttf (155.7 KB)
Extracting: D:\boot\fonts\msyh_boot.ttf (153.9 KB)
Extracting: D:\boot\fonts\msyh_console.ttf (94.2 KB)
Extracting: D:\boot\fonts\msyhn_boot.ttf (152.1 KB)
Extracting: D:\boot\fonts\segmono_boot.ttf (42.8 KB)
Extracting: D:\boot\fonts\segoe_slboot.ttf (83.2 KB)
Extracting: D:\boot\fonts\segoen_slboot.ttf (82.9 KB)
Extracting: D:\boot\fonts\wgl4_boot.ttf (46.3 KB)
Extracting: D:\boot\memtest.exe (977.3 KB)
Extracting: D:\boot\resources\bootres.dll (90.3 KB)
Extracting: D:\bootmgr (404 KB)
Extracting: D:\bootmgr.efi (1.5 MB)
```

log will go on in sec message cause too much characters for one post

weTHEeffff, 2024-03-05:
```
Extracting: D:\efi\boot\bootx64.efi (1.5 MB)
Extracting: D:\efi\microsoft\boot\bcd (16 KB)
Extracting: D:\efi\microsoft\boot\cdboot.efi (1.1 MB)
Extracting: D:\efi\microsoft\boot\cdboot_noprompt.efi (1.1 MB)
Extracting: D:\efi\microsoft\boot\efisys.bin (1.4 MB)
Extracting: D:\efi\microsoft\boot\efisys_noprompt.bin (1.4 MB)
Extracting: D:\efi\microsoft\boot\fonts\chs_boot.ttf (3.5 MB)
Extracting: D:\efi\microsoft\boot\fonts\cht_boot.ttf (3.7 MB)
Extracting: D:\efi\microsoft\boot\fonts\jpn_boot.ttf (1.9 MB)
Extracting: D:\efi\microsoft\boot\fonts\kor_boot.ttf (2.3 MB)
Extracting: D:\efi\microsoft\boot\fonts\malgun_boot.ttf (174.2 KB)
Extracting: D:\efi\microsoft\boot\fonts\malgun_console.ttf (79.5 KB)
Extracting: D:\efi\microsoft\boot\fonts\malgunn_boot.ttf (171.8 KB)
Extracting: D:\efi\microsoft\boot\fonts\meiryo_boot.ttf (141.3 KB)
Extracting: D:\efi\microsoft\boot\fonts\meiryo_console.ttf (88.5 KB)
Extracting: D:\efi\microsoft\boot\fonts\meiryon_boot.ttf (139.7 KB)
Extracting: D:\efi\microsoft\boot\fonts\msjh_boot.ttf (157.6 KB)
Extracting: D:\efi\microsoft\boot\fonts\msjh_console.ttf (198.1 KB)
Extracting: D:\efi\microsoft\boot\fonts\msjhn_boot.ttf (155.7 KB)
Extracting: D:\efi\microsoft\boot\fonts\msyh_boot.ttf (153.9 KB)
Extracting: D:\efi\microsoft\boot\fonts\msyh_console.ttf (94.2 KB)
Extracting: D:\efi\microsoft\boot\fonts\msyhn_boot.ttf (152.1 KB)
Extracting: D:\efi\microsoft\boot\fonts\segmono_boot.ttf (42.8 KB)
Extracting: D:\efi\microsoft\boot\fonts\segoe_slboot.ttf (83.2 KB)
Extracting: D:\efi\microsoft\boot\fonts\segoen_slboot.ttf (82.9 KB)
Extracting: D:\efi\microsoft\boot\fonts\wgl4_boot.ttf (46.3 KB)
Extracting: D:\efi\microsoft\boot\memtest.efi (1.3 MB)
Extracting: D:\efi\microsoft\boot\resources\bootres.dll (90.3 KB)
Extracting: D:\efi\microsoft\boot\winsipolicy.p7b (9.6 KB)
Extracting: D:\setup.exe (72.4 KB)
Extracting: D:\sources\SetupDU_166532.spdx.json (58 KB)
Extracting: D:\sources\SetupDU_166532.spdx.json.sha256 (128 bytes)
Extracting: D:\sources\acmigration.dll (364.1 KB)
Extracting: D:\sources\acres.dll (328.6 KB)
Extracting: D:\sources\actionqueue.dll (184.3 KB)
Extracting: D:\sources\adfscomp.dll (49.5 KB)
Extracting: D:\sources\admtv3check.dll (71.6 KB)
Extracting: D:\sources\aeinv.dll (755.3 KB)
Extracting: D:\sources\alert.gif (1 KB)
Extracting: D:\sources\appcompat.xsl (11.4 KB)
Extracting: D:\sources\appcompat_bidi.xsl (12.2 KB)
Extracting: D:\sources\appcompat_detailed.xsl (13.3 KB)
Extracting: D:\sources\appcompat_detailed_bidi.xsl (13.4 KB)
Extracting: D:\sources\appcompat_detailed_bidi_txt.xsl (13.1 KB)
Extracting: D:\sources\appcompat_detailed_txt.xsl (13 KB)
Extracting: D:\sources\appcompatservicing.dll (58 KB)
Extracting: D:\sources\appraiser.dll (1.8 MB)
Extracting: D:\sources\appraiser.sdb (2.5 MB)
Extracting: D:\sources\appraiserdatasha1.cat (9.3 KB)
Extracting: D:\sources\appraiserres.dll (145.1 KB)
Extracting: D:\sources\appraiserwc.dll (1.8 MB)
Extracting: D:\sources\arunimg.dll (805.8 KB)
Extracting: D:\sources\arunres.dll (13.8 KB)
Extracting: D:\sources\autorun.dll (163.8 KB)
Extracting: D:\sources\background_cli.bmp (2.3 MB)
Extracting: D:\sources\boot.wim (655.0 MB)
Extracting: D:\sources\cdplib.mof (1.9 KB)
Extracting: D:\sources\cdplibuninstall.mof (571 bytes)
Extracting: D:\sources\clustercompliance.dll (20.6 KB)
Extracting: D:\sources\cmi2migxml.dll (186.8 KB)
Extracting: D:\sources\cmisetup.dll (649.8 KB)
Extracting: D:\sources\compatctrl.dll (152.3 KB)
Extracting: D:\sources\compatresources.dll (3.6 MB)
Extracting: D:\sources\compres.dll (38.6 KB)
Extracting: D:\sources\cryptosetup.dll (23.4 KB)
Extracting: D:\sources\csiagent.dll (669.3 KB)
Extracting: D:\sources\cversion.ini (49 bytes)
Extracting: D:\sources\db_msftproductionwindowssigningca.cer (1.5 KB)
Extracting: D:\sources\devinv.dll (715.3 KB)
[... 852 lines cut ...]
Found USB 2.0 device 'USB Device' (058F:6387)
1 device found
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 949, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {2AE5A523-AAF2-4C7A-81F8-61A62A29A3B6}
Max parts: 128, Start Offset: 17408, Usable = 7811856896 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {15074BC1-757E-4D2D-A257-A8A52B3CD805}
  Size: 7.3 GB (7809757184 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {2DC620CA-56F7-4282-A2C1-DA100C412126}
  Size: 1 MB (1048576 bytes)
  Start Sector: 15255480, Attributes: 0x9000000000000000
0 devices found
0 devices found

```

weTHEeffff, 2024-03-05:
+ just tried to install it anyways but this "verified international image" does only show english.UK on install selection -__-" 
..and on former creation tries I never used "Windows User Experience options" .. but this time I selected one, it's the first time it says it uses the image selected by its name in the log ...
here a example of an other try before:  (also in 2 parts cause too long)

```
Note: No Windows User Experience options selected

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive2 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 116.8 GB)
● Creating UEFI:NTFS Partition (offset: 125458919424, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive2 for shared write access
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{32686236-589f-411b-bfcd-b6692a89a3a9}\
Disabling file indexing...
Successfully remounted \\?\Volume{32686236-589f-411b-bfcd-b6692a89a3a9}\ as F:
Extracting files...
Image is a UDF image
Extracting: F:\boot\en-us\bootsect.exe.mui (16 KB)
Extracting: F:\boot\fonts\chs_boot.ttf (3.5 MB)
Extracting: F:\boot\fonts\cht_boot.ttf (3.7 MB)
Extracting: F:\boot\fonts\jpn_boot.ttf (1.9 MB)
Extracting: F:\boot\fonts\kor_boot.ttf (2.3 MB)
Extracting: F:\boot\fonts\malgunn_boot.ttf (171.8 KB)
Extracting: F:\boot\fonts\malgun_boot.ttf (174.2 KB)
Extracting: F:\boot\fonts\malgun_console.ttf (79.5 KB)
Extracting: F:\boot\fonts\meiryon_boot.ttf (139.7 KB)
Extracting: F:\boot\fonts\meiryo_boot.ttf (141.3 KB)
Extracting: F:\boot\fonts\meiryo_console.ttf (88.5 KB)
Extracting: F:\boot\fonts\msjhn_boot.ttf (155.7 KB)
Extracting: F:\boot\fonts\msjh_boot.ttf (157.6 KB)
Extracting: F:\boot\fonts\msjh_console.ttf (198.1 KB)
Extracting: F:\boot\fonts\msyhn_boot.ttf (152.1 KB)
Extracting: F:\boot\fonts\msyh_boot.ttf (153.9 KB)
Extracting: F:\boot\fonts\msyh_console.ttf (94.2 KB)
Extracting: F:\boot\fonts\segmono_boot.ttf (42.8 KB)
Extracting: F:\boot\fonts\segoen_slboot.ttf (82.9 KB)
Extracting: F:\boot\fonts\segoe_slboot.ttf (83.2 KB)
Extracting: F:\boot\fonts\wgl4_boot.ttf (46.3 KB)
Extracting: F:\boot\resources\bootres.dll (90.3 KB)
Extracting: F:\boot\bcd (16 KB)
Extracting: F:\boot\boot.sdi (3 MB)
Extracting: F:\boot\bootfix.bin (1 KB)
Extracting: F:\boot\bootsect.exe (107.8 KB)
Extracting: F:\boot\etfsboot.com (4 KB)
Extracting: F:\boot\memtest.exe (977.3 KB)
Extracting: F:\efi\boot\bootx64.efi (1.5 MB)
Extracting: F:\efi\microsoft\boot\fonts\chs_boot.ttf (3.5 MB)
Extracting: F:\efi\microsoft\boot\fonts\cht_boot.ttf (3.7 MB)
Extracting: F:\efi\microsoft\boot\fonts\jpn_boot.ttf (1.9 MB)
Extracting: F:\efi\microsoft\boot\fonts\kor_boot.ttf (2.3 MB)
Extracting: F:\efi\microsoft\boot\fonts\malgunn_boot.ttf (171.8 KB)
Extracting: F:\efi\microsoft\boot\fonts\malgun_boot.ttf (174.2 KB)
[... 639 lines cut ...]
Extracting: F:\sources\replacementmanifests\win32k-settings-repl.man (18.3 KB)
Extracting: F:\sources\replacementmanifests\windows-senseclient-service-rep.man (1.3 KB)
Extracting: F:\sources\replacementmanifests\windows.cortana.desktop-repl.man (1.5 KB)
Extracting: F:\sources\replacementmanifests\windows.holographic.displaythrottling-replacement.man (2.6 KB)
Extracting: F:\sources\replacementmanifests\windows.media-migration-replacement.man (993 bytes)
Extracting: F:\sources\replacementmanifests\windows.sharedpc.accountmanager-repl.man (1.4 KB)
Extracting: F:\sources\replacementmanifests\windows.ui.shell-repl.man (2 KB)
Extracting: F:\sources\replacementmanifests\windowspushnotifications-platform-library-replacement.man (1.4 KB)
Extracting: F:\sources\replacementmanifests\windowssearchengine-replacement.man (6.0 KB)
Extracting: F:\sources\replacementmanifests\windowsstoreapi-replacement.man (1.1 KB)
Extracting: F:\sources\replacementmanifests\windowsupdate-replacement.man (1.6 KB)
Extracting: F:\sources\replacementmanifests\windowswebapphost-replacement.man (1.1 KB)
Extracting: F:\sources\replacementmanifests\winlogon-replacement.man (12.3 KB)
Extracting: F:\sources\replacementmanifests\winmde-migration-replacement.man (1016 bytes)
Extracting: F:\sources\replacementmanifests\winnat-replacement.man (1.2 KB)
Extracting: F:\sources\replacementmanifests\winsat-cmi-replacement.man (889 bytes)
Extracting: F:\sources\replacementmanifests\wlidsvc.vista-win7.man (4.1 KB)
Extracting: F:\sources\replacementmanifests\wmdmigration-replacement.man (4 KB)
Extracting: F:\sources\replacementmanifests\wmi-core-replacement.man (2.7 KB)
Extracting: F:\sources\replacementmanifests\wmiacpi-replacement.man (1.1 KB)
Extracting: F:\sources\replacementmanifests\wmpnsservice-migration-replacement.man (1.5 KB)
Extracting: F:\sources\replacementmanifests\wmpplayer-migration-replacement.man (2.8 KB)
Extracting: F:\sources\replacementmanifests\wsinfra-other-mw-replacement.man (1.1 KB)

```

weTHEeffff, 2024-03-05:
```

Extracting: F:\sources\replacementmanifests\wsinfra-upgrade-replacement.man (1.5 KB)
Extracting: F:\sources\replacementmanifests\wwansvc-servercfg-repl.man (1 KB)
Extracting: F:\sources\sxs\Microsoft-Windows-InternetExplorer-Optional-Package~31bf3856ad364e35~amd64~en-US~.cab (22.6 KB)
Extracting: F:\sources\sxs\microsoft-windows-internetexplorer-optional-package~31bf3856ad364e35~amd64~~.cab (285.8 KB)
Extracting: F:\sources\sxs\Microsoft-Windows-NetFx3-OnDemand-Package~31bf3856ad364e35~amd64~en-US~.cab (108.4 KB)
Extracting: F:\sources\sxs\microsoft-windows-netfx3-ondemand-package~31bf3856ad364e35~amd64~~.cab (69.4 MB)
Extracting: F:\sources\uup\metadata\desktoptargetcompdb_neutral.xml.cab (47.4 KB)
Extracting: F:\sources\vista\webservices.dll (1.1 MB)
Extracting: F:\sources\xp\webservices.dll (1.2 MB)
Extracting: F:\sources\acmigration.dll (361.5 KB)
Extracting: F:\sources\acres.dll (325.9 KB)
Extracting: F:\sources\actionqueue.dll (184.3 KB)
Extracting: F:\sources\adfscomp.dll (49.5 KB)
Extracting: F:\sources\admtv3check.dll (71.6 KB)
Extracting: F:\sources\aeinv.dll (755.3 KB)
Extracting: F:\sources\alert.gif (1 KB)
Extracting: F:\sources\appcompat.xsl (11.4 KB)
Extracting: F:\sources\appcompatservicing.dll (58 KB)Extracting: F:\sources\compatresources.dll (3.6 MB)
Extracting: F:\sources\compres.dll (38.6 KB)
Extracting: F:\sources\cryptosetup.dll (23.4 KB)
Extracting: F:\sources\csiagent.dll (709.0 KB)
Extracting: F:\sources\cversion.ini (49 bytes)
Extracting: F:\sources\db_msftproductionwindowssigningca.cer (1.5 KB)
Extracting: F:\sources\devinv.dll (715.3 KB)
Extracting: F:\sources\diager.dll (52.8 KB)
Extracting: F:\sources\diagnostic.dll (173.8 KB)
Extracting: F:\sources\diagtrack.dll (1.3 MB)
Extracting: F:\sources\diagtrackrunner.exe (86.2 KB)
Extracting: F:\sources\dism.exe (282.5 KB)
Extracting: F:\sources\dismapi.dll (1 MB)
Extracting: F:\sources\dismcore.dll (395.8 KB)
Extracting: F:\sources\dismcoreps.dll (183.5 KB)
Extracting: F:\sources\dismprov.dll (253.5 KB)
Extracting: F:\sources\du.dll (152.3 KB)
Extracting: F:\sources\facilitator.dll (1000.0 KB)
Extracting: F:\sources\folderprovider.dll (60 KB)
Extracting: F:\sources\gatherosstate.exe (1.4 MB)
Extracting: F:\sources\generaltel.dll (809 KB)
Extracting: F:\sources\hwcompat.dll (203.8 KB)
Extracting: F:\sources\hwcompat.txt (734 KB)
Extracting: F:\sources\hwcompatPE.txt (257 bytes)
Extracting: F:\sources\hwexclude.txt (51 bytes)
Extracting: F:\sources\hwexcludePE.txt (15 bytes)
Extracting: F:\sources\hypervcomplcheck.dll (182.6 KB)
Extracting: F:\sources\iasmigplugin.dll (680 KB)
Extracting: F:\sources\idwbinfo.txt (122 bytes)
Extracting: F:\sources\iiscomp.dll (23.1 KB)
Extracting: F:\sources\imagingprovider.dll (217.3 KB)
Extracting: F:\sources\input.dll (369.3 KB)
Extracting: F:\sources\install.esd (3.9 GB)
Extracting: F:\sources\itgtupg.dll (83.4 KB)
Extracting: F:\sources\lang.ini (78 bytes)
Extracting: F:\sources\locale.nls (800.3 KB)
Extracting: F:\sources\logprovider.dll (148.8 KB)
Extracting: F:\sources\mediasetupuimgr.dll (13.4 MB)
Extracting: F:\sources\migapp.xml (639.2 KB)
Extracting: F:\sources\migcore.dll (8.7 MB)
Extracting: F:\sources\mighost.exe (254.5 KB)
Extracting: F:\sources\migisol.dll (142.9 KB)
Extracting: F:\sources\migres.dll (16.9 KB)
Extracting: F:\sources\migstore.dll (1.2 MB)
Extracting: F:\sources\migsys.dll (426.0 KB)
Extracting: F:\sources\migtestplugin.dll (48.4 KB)
Extracting: F:\sources\mitigation.dll (274.9 KB)
Extracting: F:\sources\mxeagent.dll (352.0 KB)
Extracting: F:\sources\nlsbres.dll (88.5 KB)
Extracting: F:\sources\ntdsupg.dll (27.9 KB)
Extracting: F:\sources\ntfrsupg.dll (50.9 KB)
[... 165 lines cut ...]

Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15252, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {6F8188F0-7C25-4F25-9204-38801425568D}
Max parts: 128, Start Offset: 17408, Usable = 125459986944 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {32686236-589F-411B-BFCD-B6692A89A3A9}
  Size: 116.8 GB (125457870848 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {65174B39-BFB5-46B9-A070-62ED716C1E9D}
  Size: 1 MB (1048576 bytes)
  Start Sector: 245036952, Attributes: 0x9000000000000000
Saved C:\Users\Besitzer\Desktop\rufus.log

```

pbatard, 2024-03-05:
I'm afraid there's literally no way I can give you any assurance that the bootsicks are clean. If you have a virus, it could easily have injected itself after Rufus created a file, or added extra files that won't be mentioned in the log.

The only way to get the kind of assurance you seek is, after you created the boot media, to:
1. Mount the Windows ISO you used as source (make sure it is mounted read-only), preferably on a different machine, like a Linux one, and recursively create a file that contains the hashes of all the files. From Linux, you can do this with something like:  
```bash
sudo mount -o loop /home/<user>/Win11_23H2_English_x64v2.iso /mnt/iso/
find /mnt/iso -type f -exec md5sum {} \; > md5sum.txt
sudo umount /mnt/iso
```

Then you can plug your USB media, navigate to it, and use something like this to validate it:
```bash
md5sum -c /home/<user>/md5sum.txt
```

Note that you'll probably see some errors for files under `System Volume Information\` which are expected, but everything else should match.

This still will not provide you with any assurance that your target machine's UEFI firmware doesn't contain malware, in which case, even if your media passes validation, your installation of Windows will be compromises.

Since I'm afraid this is as far as I will go to try to help you with a problem that is not related to Rufus, I will close this issue.

closed 2024-03-05 by pbatard